nextcloud/nextcloudpi #1950: Apps using .mjs files (Polls) show a blank page

HansCronau, 2024-07-31:
NextCloudPi version v1.54.2
Nextcloud version 27.0.0.8
Polls version 7.1.3

After upgrading my Nextcloud to version 27 and the Polls app to 7.1.3, the Polls app now shows a blank page. It seems I am not the only one, as [an issue was filed with the Polls app](https://github.com/nextcloud/polls/issues/3561). It appears all apps using `.mjs` files (not many) are affected. Inspecting the network tab in my browser, I can indeed confirm that the file `polls-main.mjs?v=3d18ca96-5` does not load, instead returning a 404.
The issue was closed after [a solution was found](https://github.com/nextcloud/polls/issues/3561#issuecomment-2171364269). Because this solution involved changing Nginx configuration, I believe that for NCP users this might be something that should be fixed at the NCP level.

theCalcaholic, 2024-08-11:
I couldn't confirm this issue (on Nextcloud 28.0.8, NCP 1.54.2, Polls 7.2.0)

Does it still persist for you?

HansCronau, 2024-08-13:
> I couldn't confirm this issue (on Nextcloud 28.0.8, NCP 1.54.2, Polls 7.2.0)
> 
> Does it still persist for you?

It does. I haven't tested it on Nextcloud 28, though. (I am waiting with the upgrade to Nextcloud 28, until some bugs have been resolved.)

, as I'm waiting for a few bugs to be fixed before I

joshtrichards, 2025-02-04:
Full support for mjs handling was added in the default config of Nextcloud Server v28 This is no longer an issue.

Ref: nextcloud/server#38688

closed 2025-02-04 by joshtrichards