nextcloud/nextcloudpi #1952: curl and bash setup impossible either on bullseye or bookworm OS versions released by RBP Foundation

HervePruvost, 2024-08-11:
Hi,

The curl install looks broken and no more compatible with current bullseye and bookworm OS released by Rasperry Pi Foundation.

I tried on both after fresh installs from both OS, but errors are always rising.

On bullseye, I get a "distro not supported" error.

Below the console output on bullseye (both after one fresh Raspberry Pi OS image install and also after dist-upgrade just in case packages needed being upgraded):

"""
root@raspberrypi:~# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash
WARNING: This installer will disable SSH login for the root user and reset its password.
If you need to login with root, you should make sure, you have a root session open that you can use,
to revert these changes afterwards (set PermitRootLogin to 'yes' in /etc/ssh/sshd_config and run passwd as root).
Continuing in 27s (press Ctrl+C to abort)...
Continuing in 24s (press Ctrl+C to abort)...
Continuing in 21s (press Ctrl+C to abort)...
Continuing in 18s (press Ctrl+C to abort)...
Continuing in 15s (press Ctrl+C to abort)...
Continuing in 12s (press Ctrl+C to abort)...
Continuing in 9s (press Ctrl+C to abort)...
Continuing in 6s (press Ctrl+C to abort)...
Continuing in 3s (press Ctrl+C to abort)...
Continuing in 0s (press Ctrl+C to abort)...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:4 http://deb.debian.org/debian bullseye-updates InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ca-certificates is already the newest version (20210119).
git is already the newest version (1:2.30.2-1+deb11u2).
gnupg2 is already the newest version (2.2.27-2+deb11u2).
jq is already the newest version (1.6-2.1).
lsb-release is already the newest version (11.1.0).
sudo is already the newest version (1.9.5p2-3+deb11u1).
wget is already the newest version (1.21-1+deb11u1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Getting build code...
Cloning into '/tmp/nextcloudpi.iCfH62/nextcloudpi'...
remote: Enumerating objects: 16294, done.
remote: Counting objects: 100% (2998/2998), done.
remote: Compressing objects: 100% (580/580), done.
remote: Total 16294 (delta 2607), reused 2601 (delta 2415), pack-reused 13296
Receiving objects: 100% (16294/16294), 8.91 MiB | 5.59 MiB/s, done.
Resolving deltas: 100% (11484/11484), done.

Installing NextCloudPi...
etc/library.sh: line 17: php: command not found
ERROR: distro not supported:
Debian GNU/Linux 11 \n \l
----------------------------------------------

Trying a curl install after installation of bookworm OS (both fresh from raspi imager and after dist-upgrade), I basically got the same issues pointed out in #1926, #1911, #1663.

More concretely, following issues raised:
- the "maxmemory invalid argument" from redis config file, which I had to set manually (e.g. at 512mb)
- the "PHP Warning: Undefined variable $CONFIG in Command line code line1. But it does not break the install.
- and the issue with the missing systemd-resolved package which I had to install manually.

At the end, it looked NCP had been installed, but ending the install.sh script by "Set Admin Password" output, and never outputting "Done" that means the end of the script.

Trying then to activate my NCP setup on "https://myIP/", clicking on activate raised a "NextCloudPi not yet initialized, trying again in a few seconds" loop. But NCP never activated and the setup failed.

As a conclusion, it looks there is currently no way of making a successful install on preinstalled existing Debian OS from the Raspberry Pi Foundation.

Please try fixing the .sh install scripts, cause this is very disabling for people using officlal RBP OS versions.

theCalcaholic, 2024-08-12:
Duplicate of #1926

theCalcaholic, 2024-08-12:
> Duplicate of #1926

Actually, I have to correct myself: you're running into a different error, because you seem to be installing NCP on an outdated base distro (Debian 11 is not support anymore).

Please use an up to date debian 12 system or, even better, one of the provided ready-to-install images: https://github.com/nextcloud/nextcloudpi/releases

closed 2024-08-12 by theCalcaholic