nextcloud/nextcloudpi #1956: Could not find a valid upgrade path from '28.0.8'

Tommy-01, 2024-08-18:
**System information**
Nextcloudpi runnig in "Docker Station" an QNAP NAS.

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.54.3
NextcloudPi image    NextCloudPi_LXD_11-15-23.img
OS                   Debian GNU/Linux 11. 5.10.60-qnap (x86_64)
automount            no
USB devices          sdc 
datadir              /opt/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.1T/3.6T
rootfs usage         2.1T/3.6T
swapfile             /share/CACHEDEV1_DATA/.swap/qnap_swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    28.0.8.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               9days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "NextcloudLXD2",
            "11": "91.15.80.149",
            "1": "192.168.178.43",
            "14": "NextcloudLXD2"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "28.0.8.1",
        "overwrite.cli.url": "https:\/\/NextcloudLXD2\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sun Aug 18 20:04:17.434699 2024] [proxy_fcgi:error] [pid 626:tid 636] [remote 192.168.178.30:37540] AH01067: Failed to read FastCGI header
[Sun Aug 18 20:04:17.455462 2024] [proxy_fcgi:error] [pid 626:tid 636] (104)Connection reset by peer: [remote 192.168.178.30:37540] AH01075: Error dispatching request to :4443: 
[Sun Aug 18 20:04:18.141381 2024] [proxy_fcgi:error] [pid 626:tid 658] [remote 192.168.178.30:37540] AH01067: Failed to read FastCGI header
[Sun Aug 18 20:04:18.141448 2024] [proxy_fcgi:error] [pid 626:tid 658] (104)Connection reset by peer: [remote 192.168.178.30:37540] AH01075: Error dispatching request to :4443: 
[Sun Aug 18 20:30:36.667132 2024] [proxy:error] [pid 626:tid 642] (103)Software caused connection abort: [remote 192.168.178.30:41450] AH01095: prefetch request body failed to ?:0 (localhost) from 192.168.178.30 (blue-huawei.fritz.box)
[Sun Aug 18 20:33:51.864975 2024] [authnz_external:error] [pid 626:tid 648] [remote 192.168.178.30:40460] AuthExtern pwauth [/usr/sbin/pwauth]: Fai
```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairInfoEvent: Repair info: No need to remove link shares.","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Clear access cache of projects","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Reset generated avatar flag","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+00:00","remoteAddr":"","user":"--","app":"updater",{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Remove old dashboard app config data","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Add job to cleanup the bruteforce entries","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Queue a one-time job to check for user uploaded certificates","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:42+0:maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"28.0.8.1","data":{"app":"updater"}}
{"reqId":"1HOigncXCFf5fLCGq4Mv","level":1,"time":"2024-08-18T20:03:55+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"28.0.8.1","data":{"app":"updater"}}

```
</details>

Problem:
Running  "Update NextCloudPi" on the NextcloudPi Panel in "Container Station"  results in:

```
Update NextCloudPi

[ nc-update ] (Sun Aug 18 20:55:34 UTC 2024)
Downloading updates
Performing updates
Update to bookworm available. Type 'sudo ncp-dist-upgrade' to upgrade

```
Running 'sudo ncp-dist-upgrade' in the Console of "Container Station" results in:
```
root@NextcloudLXD2:~# sudo ncp-dist-upgrade
no server running on /tmp/tmux-0/default
open terminal failed: missing or unsuitable terminal: unknown
```
Shouldn't the update only be done via the NextcloudPi panel? But unfortunately this currently only returns an error message, see above.

theCalcaholic, 2024-08-18:
Is there something missing from the output of "Update NextcloudPi" (like the error message from the title of the issue)?

Tommy-01, 2024-08-19:
Ah sry, the Error Message from "nc-update-nextcloud" was 
```
Could not find a valid upgrade path from '28.0.8'".
```

Today it says 
```
[ nc-update-nextcloud ] (Mon Aug 19 19:23:48 UTC 2024)
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed

Nextcloud version 28.0.8 is already installed. Nothing to do.
```

I don't know why it didn't say yesterday that it was up to date.
I think that's fine and the ticket can be closed.

theCalcaholic, 2024-08-19:
> I don't know why it didn't say yesterday that it was up to date.

Because I fixed the status message yesterday after you reported it :)

closed 2024-08-19 by theCalcaholic