nextcloud/nextcloudpi #1958: Internal Server Error and HBP Error

rustyb78, 2024-08-21:
Internal Server Error and HBP Error:

After 1.54.3 was installed my instance is having issues. I have posted in the forum but I'm also posting here.

Thanks.

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.54.3
NextcloudPi image    NextCloudPi_RaspberryPi_v1.50.3.img
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      3.3T/3.7T
rootfs usage         7.5G/29G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    28.0.8.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              [N/A]
Port 443             [N/A]
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               7:38

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "ncp.rustybryant.com",
            "11": "<!DOCTYPE HTML PUBLIC \"-\/\/IETF\/\/DTD HTML 2.0\/\/EN\">\n<html><head>\n<title>302 Found<\/title>\n<\/head><body>\n<h1>Found<\/h1>\n<p>The document has moved <a href=\"https:\/\/icanhazip.com\/\">here<\/a>.<\/p>\n<\/body><\/html>",
            "1": "192.168.1.85",
            "14": "nextcloudpi",
            "20": "ncp.bradleycountytn.com",
            "21": "ncp.rustybryant.com",
[... 72 lines cut ...]
[Wed Aug 21 17:45:12.236187 2024] [authz_host:error] [pid 1712:tid 1715] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:12.409438 2024] [authz_host:error] [pid 1712:tid 1763] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /img/favicon.png failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:19.739314 2024] [authz_host:error] [pid 1712:tid 1779] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:19.913838 2024] [authz_host:error] [pid 1712:tid 1779] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:19.957981 2024] [authz_host:error] [pid 1712:tid 1753] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:19.963927 2024] [authz_host:error] [pid 1712:tid 1788] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /img/download.svg failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:19.967884 2024] [authz_host:error] [pid 1712:tid 1725] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /img/delete.svg failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:20.178406 2024] [authz_host:error] [pid 1712:tid 1725] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /img/delete.svg failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:20.184047 2024] [authz_host:error] [pid 1712:tid 1753] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Wed Aug 21 17:45:20.186199 2024] [authz_host:error] [pid 1712:tid 1788] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /img/download.svg failed, reason: unable to get the remote host name
[Wed Aug 21 17:47:04.853922 2024] [authz_host:error] [pid 1712:tid 1750] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /img/info.svg failed, reason: unable to get the remote host name, referer: https://192.168.1.85:4443/css/ncp.css
[Wed Aug 21 17:47:05.038573 2024] [authz_host:error] [pid 1712:tid 1750] [remote 192.168.1.1:1439] AH01753: access check of 'localhost' to /img/info.svg failed, reason: unable to get the remote host name, referer: https://192.168.1.85:4443/css/ncp.css

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

theCalcaholic, 2024-08-24:
Can you post the output of `sudo tail -n 50 /media/myCloudDrive/ncdata/data` please (check it for sensitive data first, but it probably shouldn't include any).

Ideally, use a service like https://bin.disroot.org/ to post the output

rustyb78, 2024-08-25:
> Can you post the output of `sudo tail -n 50 /media/myCloudDrive/ncdata/data` please (check it for sensitive data first, but it probably shouldn't include any).
> 
> Ideally, use a service like https://bin.disroot.org/ to post the output

pi@nextcloudpi:~ $ sudo tail -n 50 /media/myCloudDrive/ncdata/data
tail: error reading '/media/myCloudDrive/ncdata/data': Is a directory

rustyb78, 2024-08-25:
I dug down to the log file. 
https://bin.disroot.org/?56c00d9a069dfdb6#7TeuHXNMzeJmpSYRVCA4JTGyqNoJAGvLRhZFJw771gzv

theCalcaholic, 2024-08-25:
Looks like your database is broken(it's missing tables that should be there). You will need to restore from backup.

closed 2024-08-25 by theCalcaholic