nextcloud/nextcloudpi #1959: Include a database dump in the data dir before doing a btrfs-snapshot

bearsh, 2024-08-22:
As far as I know, the `nc-snapshot` script does not backup the database. The `nc-backup-auto` script is able to keep a configurable number of backups but, as these backups are located outside of the ncdata directory, the `nc-snapshot-sync` script won't be able to send them to another device. Furthermore the algorithm for keeping a number of old backups is quite simple (only a number).

So my proposal is:
- before doing a btrfs snapshot of the ncdata directory, call the `nc-backup` script (exclude data) with a destination path inside the ncdata directory.
- make the period of the `nc-snapshot-auto` script configurable, I doubt everyone would need a backup for every hour.

This results in snapshots which also includes the database dump which makes the backup handling much easier as the snapshot algorithm already takes care of keeping older snapshot in a intelligent way. furthermore that dump will also end up on a remote machine if `nc-snapshot-sync` is used.

bearsh, 2024-08-22:
> You would also need to create a backup of the nextcloud directory and copy that into the data directory every time, because a database snapshot needs the exact version of NC and installed apps to be guaranteed to be restorable without issues.

is this not already done? my ncdata directory contains a folder ncp-update-backups with all (not only the latest few :thinking: ) nextcloud updates: like `nextcloud-bkp_20190707_1562480891-15.0.8.1.tar.gz` or `nextcloud-bkp_20240510_1715322466-28.0.4.1.tar.gz`

> In the case of my instance, that's roughly 16GB of data in total (about 500MB from the db and 15,5GB from the nextcloud data directory due to a few larger apps).
This means, instead of no additional disk space, every snapshot would cost 16GB every hour.

I don't get this. if only the db dump changes every hour, the cost is only the size of the dump file...

> Because of these issues and also, because I will soon add a new backup method (based on https://kopia.io) ...

that sounds very promising! and I fully understand you want to focus on one solution :+1:

theCalcaholic, 2024-08-22:
Yes, the nextcloud-*.tar.gz files contain both the database as well as the nextcloud directory, to be able to restore your system. The same would be necessary for the dumps included with the btrfs snapshots.

theCalcaholic, 2024-08-22:
My point is: To be sure that you can successfully restore a db backup, you need to also make a backup of the nextcloud directory every time you perform the database backup (app versions might have changed that include database changes).

closed 2024-08-22 by theCalcaholic