nextcloud/nextcloudpi #1960: nc-update-nextcloud not regonzizing debian 12

stoneubi, 2024-08-23:
Hi there,

my setup is:
* proxmox server on latest version 8.2.4
* OS is dietpi 9.6.1 debian 12 bookworm for proxmox 
* ncp version 1.54.3
* nextcloud version is 29.0.4

Using nc-update-nextcloud and manually entering to update to version 29.0.5. leads to the following information: 
* ncp doesn't support Nextcloud version greater than 28 with Debian 11 (Bullseye). Please run ncp-dist-upgrade.

[image]

Running ncp-dist-upgrade results in:
* No dist-upgrade available for OS version: Debian 12 (bookworm). 

[image]

Till today, I used nc-update-nextcloud to manually update to the latest version. I would like to update to 29.0.5 and ask for support to be able to do this again. 

Many thanks,
Stoneubi

theCalcaholic, 2024-08-23:
That means that your dist upgrade didn't complete successfully. See #1913 for instructions for manually competing it.

stoneubi, 2024-08-23:
Hi, 

thank you very much for your prompt reply. 

I followed #1913 and checked if any bullseye entries are available. There where no entries. All entries are on bookworm. 

I also executed all mentioned steps and rebooted the server. 

Anyhow - same result. Updating to 29.0.4 went fine a few week before.  Updating to 29.0.5 still not starts because of above mentioned message: ncp doesn't support Nextcloud version greater than 28 with Debian 11 (Bullseye). Please run ncp-dist-upgrade.

Do you have any other recommendations? 

Many thanks,
Stoneubi

theCalcaholic, 2024-08-23:
Oh you're right, apparently I introduced a bug with the latest release. It will be fixed soon.

The next release will also bring an automatic upgrade to PHP 8.3 when you update Nextcloud, so it might be worthwhile to wait with that until the end of the week.

theCalcaholic, 2024-09-07:
Fixed in v1.55.0

closed 2024-09-07 by theCalcaholic

stoneubi, 2024-09-14:
Thank you :-) Now it works.