nextcloud/nextcloudpi #1963: Error while applying update 1.54.0

Lucae81110, 2024-08-23:
Hello,
My Nextcloud is running fine with V1.53.3 but I've an issue while trying to update ncp from NextcloudP i v1.53.3 i to v1.54.3
I've this report:
<details>
[ nc-update ] (Fri Aug 23 15:05:31 CEST 2024)
Downloading updates
Performing updates
Update root login prevention method...
done.
Fixing trusted proxies list...
done.
Updating PHP package signing key...
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [39.0 kB]
Get:5 https://packages.sury.org/php bullseye InRelease [7551 B]
Hit:6 http://deb.debian.org/debian bullseye-updates/main armhf Packages
Hit:7 http://deb.debian.org/debian bullseye-updates/main arm64 Packages
Hit:8 http://deb.debian.org/debian bullseye-updates/main Translation-en
Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [311 kB]
Get:10 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [318 kB]
Err:5 https://packages.sury.org/php bullseye InRelease
The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key
Fetched 719 kB in 1s (483 kB/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php bullseye InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key
W: Problem unlinking the file /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease - TransactionStates-Cleanup (21: Is a directory)
W: Failed to fetch https://packages.sury.org/php/dists/bullseye/InRelease The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key
W: Some index files failed to download. They have been ignored, or old ones used instead.
E: Could not read from /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease - getline (21: Is a directory)
E: The package lists or status file could not be parsed or opened.
Error while applying update 1.54.0. Exiting...
</details>

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.53.3
NextcloudPi image    NextCloudPi_RaspberryPi_v1.52.0.img
OS                   Debian GNU/Linux 11. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda sdb sdc 
datadir              /media/SSD-1/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      2.2G/477G
rootfs usage         6.4G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    26.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            wlan0
[... 91 lines cut ...]
[Fri Aug 23 16:01:27.221614 2024] [authz_host:error] [pid 1220238:tid 548449620352] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /img/logs.svg failed, reason: unable to get the remote host name, referer: https://192.168.1.50:4443/css/ncp.css
[Fri Aug 23 16:01:27.226316 2024] [authz_host:error] [pid 1220238:tid 548432802176] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /img/info-white.svg failed, reason: unable to get the remote host name, referer: https://192.168.1.50:4443/css/ncp.css
[Fri Aug 23 16:01:27.245195 2024] [authz_host:error] [pid 1220238:tid 548491641216] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /img/search.svg failed, reason: unable to get the remote host name, referer: https://192.168.1.50:4443/css/ncp.css
[Fri Aug 23 16:01:27.247167 2024] [authz_host:error] [pid 1220238:tid 548441211264] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /ncp-output.php failed, reason: unable to get the remote host name
[Fri Aug 23 16:01:27.265314 2024] [authz_host:error] [pid 1220238:tid 548466438528] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /img/defaults-white.svg failed, reason: unable to get the remote host name, referer: https://192.168.1.50:4443/css/ncp.css
[Fri Aug 23 16:01:27.276547 2024] [authz_host:error] [pid 1220238:tid 548323484032] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /img/wizard.svg failed, reason: unable to get the remote host name, referer: https://192.168.1.50:4443/css/ncp.css
[Fri Aug 23 16:01:27.526202 2024] [authz_host:error] [pid 1220238:tid 548373938560] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /img/favicon.png failed, reason: unable to get the remote host name
[Fri Aug 23 16:01:27.716884 2024] [authz_host:error] [pid 1220238:tid 548373938560] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /img/favicon.png failed, reason: unable to get the remote host name
[Fri Aug 23 16:01:37.827597 2024] [authz_host:error] [pid 1220238:tid 548474847616] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Fri Aug 23 16:01:38.003065 2024] [authz_host:error] [pid 1220238:tid 548474847616] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Fri Aug 23 16:01:38.018679 2024] [authz_host:error] [pid 1220238:tid 548483248512] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Fri Aug 23 16:01:38.211542 2024] [authz_host:error] [pid 1220238:tid 548483248512] [remote 192.168.1.17:59622] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

theCalcaholic, 2024-08-23:
Hi, seems like you're running into an older issue, that should be fixed on a later version but prevents you to update in order to get the fix. Running the following commands from a terminal should resolve your issue (it's the same fix that would have been applied by a later update:

```sh
sudo apt-get update
sudo apt-get install --no-install-recommends -y gnupg2
sudo apt-key adv --fetch-keys https://packages.sury.org/php/apt.gpg
```

Lucae81110, 2024-08-24:
Hello,
Thanks for the answer. But I' still have some errors.
<details>
<summary>When I run sudo apt-get update </summary>
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease                          
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]        
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease                 
Get:5 https://packages.sury.org/php bullseye InRelease [7,551 B]               
Hit:6 http://deb.debian.org/debian bullseye-updates/main armhf Packages
Hit:7 http://deb.debian.org/debian bullseye-updates/main arm64 Packages
Hit:8 http://deb.debian.org/debian bullseye-updates/main Translation-en
Err:5 https://packages.sury.org/php bullseye InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Fetched 7,551 B in 1s (5,253 B/s)
Reading package lists... Error!
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php bullseye InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Problem unlinking the file /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease - TransactionStates-Cleanup (21: Is a directory)
W: Failed to fetch https://packages.sury.org/php/dists/bullseye/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
E: Could not read from /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease - getline (21: Is a directory)
E: The package lists or status file could not be parsed or opened.
</details>

<details>
<summary>When I run sudo apt-get install --no-install-recommends -y gnupg2</summary>
Reading package lists... Error!
E: Could not read from /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease - getline (21: Is a directory)
E: The package lists or status file could not be parsed or opened.
</details>

<details>
<summary>When I run sudo apt-key adv --fetch-keys https://packages.sury.org/php/apt.gpg</summary>
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.m1m13ofy4S/gpg.1.sh --fetch-keys https://packages.sury.org/php/apt.gpg
gpg: requesting key from 'https://packages.sury.org/php/apt.gpg'
gpg: key B188E2B695BD4743: "DEB.SURY.ORG Automatic Signing Key <deb@sury.org>" 2 new signatures
gpg: Total number processed: 1
gpg:         new signatures: 2
</details>

<details>
<summary>When I run nc-update</summary>
[ nc-update ] (Sat Aug 24 07:38:52 CEST 2024)
Downloading updates
Performing updates
Update root login prevention method...
done.
Fixing trusted proxies list...
done.
Updating PHP package signing key...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Get:5 https://packages.sury.org/php bullseye InRelease [7551 B]
Hit:6 http://deb.debian.org/debian bullseye-updates/main armhf Packages
Hit:7 http://deb.debian.org/debian bullseye-updates/main arm64 Packages
Hit:8 http://deb.debian.org/debian bullseye-updates/main Translation-en
Get:9 https://packages.sury.org/php bullseye/main arm64 Packages [248 kB]
Get:10 https://packages.sury.org/php bullseye/main armhf Packages [248 kB]
Fetched 496 kB in 2s (263 kB/s)
Reading package lists...

W: Problem unlinking the file /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease - TransactionStates-Cleanup (21: Is a directory)
E: Could not read from /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease - getline (21: Is a directory)
E: The package lists or status file could not be parsed or opened.
Error while applying update 1.54.0. Exiting...
</details>

theCalcaholic, 2024-08-24:
Ok, looks like your apt source configuration has an invalid file for some reason. What's the output of the following?

```sh
set -x && sudo ls -l /etc/apt/sources.list* /var/lib/apt/lists/ && sudo cat /etc/apt/sources.list.d/*; set +x
```

Please enclose any code in `\`\`\`` to make it readable, like this:

    ```
    script output here
    ```

Lucae81110, 2024-08-24:
This is the out put

```
+ sudo ls -l /etc/apt/sources.list /etc/apt/sources.list.d /var/lib/apt/lists/
-rw-r--r-- 1 root root  542 Sep 22  2022 /etc/apt/sources.list

/etc/apt/sources.list.d:
total 8
-rw-r--r-- 1 root root  49 May  1  2023 php.list
-rw-r--r-- 1 root root 191 Sep 22  2022 raspi.list

/var/lib/apt/lists/:
total 129252
-rw-r--r-- 1 root root    38985 Aug 23 15:31 archive.raspberrypi.org_debian_dists_bullseye_InRelease
-rw-r--r-- 1 root root  1212064 Aug 23 15:31 archive.raspberrypi.org_debian_dists_bullseye_main_binary-arm64_Packages
-rw-r--r-- 1 root root  1233359 Aug 23 15:31 archive.raspberrypi.org_debian_dists_bullseye_main_binary-armhf_Packages
drwxr-xr-x 2 _apt root     4096 Sep  2  2023 auxfiles
-rw-r--r-- 1 root root   181252 Jun 29 11:50 deb.debian.org_debian_dists_bullseye_contrib_binary-arm64_Packages
-rw-r--r-- 1 root root   175958 Apr 29  2023 deb.debian.org_debian_dists_bullseye_contrib_binary-armhf_Packages
-rw-r--r-- 1 root root   192685 Jul  9  2022 deb.debian.org_debian_dists_bullseye_contrib_i18n_Translation-en
-rw-r--r-- 1 root root   115938 Jun 29 12:38 deb.debian.org_debian_dists_bullseye_InRelease
-rw-r--r-- 1 root root 44839134 Jun 29 11:51 deb.debian.org_debian_dists_bullseye_main_binary-arm64_Packages
-rw-r--r-- 1 root root 43914977 Jun 29 11:51 deb.debian.org_debian_dists_bullseye_main_binary-armhf_Packages
-rw-r--r-- 1 root root 30255023 Jun 29 11:51 deb.debian.org_debian_dists_bullseye_main_i18n_Translation-en
-rw-r--r-- 1 root root   379784 Jun 29 11:50 deb.debian.org_debian_dists_bullseye_non-free_binary-arm64_Packages
-rw-r--r-- 1 root root   259198 Jun 29 11:50 deb.debian.org_debian_dists_bullseye_non-free_binary-armhf_Packages
-rw-r--r-- 1 root root   565058 Jun 29 11:50 deb.debian.org_debian_dists_bullseye_non-free_i18n_Translation-en
drwxrwxrwt 2 root root     4096 Sep 23  2023 deb.debian.org_debian_dists_bullseye-updates_InRelease
-rw-r--r-- 1 root root    59890 Dec 29  2023 deb.debian.org_debian_dists_bullseye-updates_main_binary-arm64_Packages
-rw-r--r-- 1 root root    60588 Dec 29  2023 deb.debian.org_debian_dists_bullseye-updates_main_binary-armhf_Packages
-rw-r--r-- 1 root root    58637 Dec 29  2023 deb.debian.org_debian_dists_bullseye-updates_main_i18n_Translation-en
-rw-r----- 1 root root        0 Sep  2  2023 lock
-rw-r--r-- 1 root root     7551 Aug 24 07:40 packages.sury.org_php_dists_bullseye_InRelease
-rw-r--r-- 1 root root  1962979 Aug  6 11:01 packages.sury.org_php_dists_bullseye_main_binary-arm64_Packages
-rw-r--r-- 1 root root  1958965 Aug  6 10:00 packages.sury.org_php_dists_bullseye_main_binary-armhf_Packages
drwx------ 2 _apt root     4096 Aug 24 07:40 partial
-rw-r--r-- 1 root root    48377 Aug 23 18:25 security.debian.org_debian-security_dists_bullseye-security_InRelease
-rw-r--r-- 1 root root  1755569 Aug 13 08:47 security.debian.org_debian-security_dists_bullseye-security_main_binary-arm64_Packages
-rw-r--r-- 1 root root  1762265 Aug 13 08:47 security.debian.org_debian-security_dists_bullseye-security_main_binary-armhf_Packages
-rw-r--r-- 1 root root  1245855 Aug 12 20:28 security.debian.org_debian-security_dists_bullseye-security_main_i18n_Translation-en
-rw-r--r-- 1 root root      924 Jul 25  2023 security.debian.org_debian-security_dists_bullseye-security_non-free_i18n_Translation-en
+ sudo cat /etc/apt/sources.list.d/php.list /etc/apt/sources.list.d/raspi.list
deb https://packages.sury.org/php/ bullseye main
deb http://archive.raspberrypi.org/debian/ bullseye main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ bullseye main
+ set +x

```

theCalcaholic, 2024-08-24:
Can you `sudo mv  /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease /tmp/`? I don't think that directory is supposed to be there.

After that run
```sh
sudo apt-get update
sudo apt-get install --no-install-recommends -y gnupg2
sudo apt-key adv --fetch-keys https://packages.sury.org/php/apt.gpg
```
again

Lucae81110, 2024-08-25:
After running the code I've this
```
sudo mv  /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease /tmp/
mv: cannot move '/var/lib/apt/lists/deb.debian.org_debian_dists_bullseye-updates_InRelease' to '/tmp/deb.debian.org_debian_dists_bullseye-updates_InRelease': Structure needs cleaning
```

theCalcaholic, 2024-08-25:
Unfortunately, that sounds like a disk/filesystem issue. You will need to check the health of your disk and maybe run fsck on your filesystem.

closed 2024-08-25 by theCalcaholic