nextcloud/nextcloudpi #1965: Daily update notification of new nc version since last update from 1.54.2 to 1.54.3

CarlosCastaneda, 2024-09-01:
**System information**

Since the latest update to ncp version 1.54.3 a daily notification of
```
NextCloudPi
Nextcloud was updated to 29.0.5
```
even though nc is at the last update level and nothing has changed.

gillesF78, 2024-09-06:
I encounter the same issue.

theCalcaholic, 2024-09-07:
This should be fixed with v1.55.0

gillesF78, 2024-09-10:
My NCP was updated to v1.55.1, and it's keeping sending this false daily update notification.

CarlosCastaneda, 2024-09-10:
> My NCP was updated to v1.55.1, and it's keeping sending this false daily update notification.

Yep, no changes in behaviour so far. Not in targeted version 1.55.0 nor in 1.55.1.

theCalcaholic, 2024-09-24:
Should be fixed in v1.55.2 (the issues within the last versions were actually slightly different but all related to the ncp-update-nc overhault)

closed 2024-09-24 by theCalcaholic