nextcloud/nextcloudpi #1967: nc_update does not work

janvlug, 2024-09-06:
When I run nc_update from the web interface, it seems like an update starts, I see:

[image]

On the CLI, I see that `update.sh` is running in `top`:

```
1772068 root      20   0    4188   3204   2504 S   0.7   0.0   0:00.39 update.sh                                                                                                              
```

However, it seems that the update does not finalize, but results in a timeout or something like that. After waiting for some time, I see: `Something went wrong. Try refreshing the page` in the web interface, then it seems that the browser 'hangs' on loading the page, and after waiting even longer I get redirected to the login page.

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.54.3
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.3.img
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/nextcloudpi_data/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      2.7T/5.3T
rootfs usage         7.5G/59G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    28.0.9.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               24days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "<redacted>",
            "11": "2001:1c00:c0b:8a00:d8ba:2e77:2f5b:472",
[... 79 lines cut ...]
[Fri Sep 06 10:59:26.659543 2024] [authz_host:error] [pid 1560304:tid 1560337] [remote 192.168.178.129:46104] AH01753: access check of 'localhost' to /img/delete.svg failed, reason: unable to get the remote host name

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

<redacted>
{"reqId":"ZtrG78q4EEjePHAxKLst8QAAFxc","level":3,"time":"2024-09-06T09:10:07+00:00","remoteAddr":"83.87.39.90","user":"<redacted>","app":"no app in context","method":"GET","url":"/ocs/v2.php/apps/dashboard/api/v2/widget-items?widgets%5B%5D=mail","message":"Exception thrown: Symfony\\Component\\Routing\\Exception\\InvalidParameterException","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:129.0) Gecko/20100101 Firefox/129.0","version":"28.0.9.1","exception":{"Exception":"Symfony\\Component\\Routing\\Exception\\InvalidParameterException","Message":"Parameter \"guestName\" for route \"core.GuestAvatar.getAvatar\" must match \"[^/]++\" (\"\" given) to generate a corresponding URL.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/symfony/routing/Generator/UrlGenerator.php","line":161,"function":"doGenerate","class":"Symfony\\Component\\Routing\\Generator\\UrlGenerator","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":372,"function":"generate","class":"Symfony\\Component\\Routing\\Generator\\UrlGenerator","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/CachingRouter.php","line":65,"function":"generate","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/lib/private/URLGenerator.php","line":103,"function":"generate","class":"OC\\Route\\CachingRouter","type":"->"},{"file":"/var/www/nextcloud/apps/mail/lib/Dashboard/MailWidget.php","line":167,"function":"linkToRoute","class":"OC\\URLGenerator","type":"->"},{"function":"OCA\\Mail\\Dashboard\\{closure}","class":"OCA\\Mail\\Dashboard\\MailWidget","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/apps/mail/lib/Dashboard/MailWidget.php","line":152,"function":"array_map"},{"file":"/var/www/nextcloud/apps/mail/lib/Dashboard/MailWidgetV2.php","line":43,"function":"getItems","class":"OCA\\Mail\\Dashboard\\MailWidget","type":"->"},{"file":"/var/www/nextcloud/apps/dashboard/lib/Controller/DashboardApiController.php","line":142,"function":"getItemsV2","class":"OCA\\Mail\\Dashboard\\MailWidgetV2","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":230,"function":"getWidgetItemsV2","class":"OCA\\Dashboard\\Controller\\DashboardApiController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":137,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":184,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":315,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/ocs/v1.php","line":65,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/ocs/v2.php","line":23,"args":["/var/www/nextcloud/ocs/v1.php"],"function":"require_once"}],"File":"/var/www/nextcloud/3rdparty/symfony/routing/Generator/UrlGenerator.php","Line":194,"CustomMessage":"Exception thrown: Symfony\\Component\\Routing\\Exception\\InvalidParameterException"}}

```
</details>

theCalcaholic, 2024-09-07:
Can you try again after the update to v1.55.0 and see if the issue persists?

janvlug, 2024-09-08:
I upgraded to v1.55.0, and after that did the nextcloud update from the web interface. All went well. I am now on Nextcloud 29.0.4.

And if I try to upgrade Nextcloud now from the web interface, I get:

```
[ nc-update-nextcloud ] (Sun Sep 8 12:59:49 BST 2024)
Failed to reset failed state of unit ncp-encrypt.service: Unit ncp-encrypt.service not loaded.
Running as unit: ncp-update-nc.service
Nextcloud version 29.0.4 is already installed. Nothing to do.
Nextcloud update finished successfully.
```

Which seems good to me. So closing this issue.

closed 2024-09-08 by janvlug