nextcloud/nextcloudpi #1969: Need assistance upgrading to Nextcloud 29

Haraade, 2024-09-07:
Need assistance upgrading to Nextcloud 29.0.x. I haven't installed and used the social app in Nextcloud, but during upgrade, installation is interrupted due to: InvalidArgumentException: Index name "nid" for table "oc_social_3_cache_actor" collides with the constraint on table "oc_social_3_stream". As I see it, social must be removed from the database. How to perform this with NextcloudPi?

theCalcaholic, 2024-09-07:
Hi, this is not really related to NextcloudPi, as it is a general DB issue with an app.
You can ask for help in the [Nextcloud help forum](https://help.nextcloud.com) or ask the author of the Nextcloud app in question.

I myself can't offer support for individual instances unless you would like to hire me for paid freelance work.

closed 2024-09-07 by theCalcaholic

Haraade, 2024-09-08:
Thank you for your assistance. For others who would have this problem.

mysql -u ncadmin --password=$MYSQL_PASSWORD nextcloud -e "show tables" -s | grep "social" | xargs -I "@@" mysql -u ncadmin --password=$MYSQL_PASSWORD nextcloud -e "DROP TABLE @@"