nextcloud/nextcloudpi #1970: Can't set Wireless LAN (raspi-config - Pi4B)

Godrigos, 2024-09-08:
I tried in two fresh installs and after activation, basic configuration and enabling SSH, when I use `raspi-config` to set my wifi I get the following error:

`
Traceback (most recent call last):
  File "/usr/sbin/netplan", line 23, in <module>
    netplan.main()
  File "/usr/share/netplan/netplan/cli/core.py", line 50, in main
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 50, in run
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 72, in command_set
    libnetplan.create_yaml_patch(yaml_path, value, tmp)
  File "/usr/share/netplan/netplan/libnetplan.py", line 501, in create_yaml_patch
    _checked_lib_call(lib.netplan_util_create_yaml_patch,
  File "/usr/share/netplan/netplan/libnetplan.py", line 79, in _checked_lib_call
    raise LibNetplanException(err.contents.message.decode('utf-8'))
netplan.libnetplan.LibNetplanException: Error parsing YAML: did not find expected ',' or '}'
`

And no wifi connection.

System data:
NextcloudPi v1.55.0
Raspberry Pi 4B 8 gb
Linux nextcloudpi 6.6.45-current-bcm2711 #2 SMP PREEMPT Fri Aug 16 20:53:33 UTC 2024 aarch64 GNU/Linux

Peppeeeerrrr, 2024-09-18:
Hi, I had the same problem here with an RP4B too. I installed the older version of ncp v1.54.0 as the wifi seemed to work back then. The configuration through  `raspi-config ` didn't work out neither but  I managed to activate the wifi installing and running `armbian-config`. Seems a good way to get around the issue.

Godrigos, 2024-09-18:
Thanks for the tip. Will try it later.

theCalcaholic, 2024-09-19:
@Peppeeeerrrr Thank you, seems like a good workaround.

The issue itself is very likely an upstream issue with Armbian for the RPi.

@Godrigos If you are able to confirm that the workaround works for you, we can close this issue (and I'll ask the Wiki team to add it to the NCP wiki).

Godrigos, 2024-09-19:
Unfortunatly, even if I can install the armbian-config, it does not present options for wifi configurations, there is a warning that missing drivers may limit the configuration options.

Godrigos, 2024-09-21:
I solved my isseu with an ugly long cable, lol. But no wifi.
Thanks for the attention guys.

andrei7c4, 2024-12-28:
> Sad. I need wifi acces too. I dont want use cable, coz this pi4 is hiddne uder bed :D Solved. Install network-manager and Wifi is enabled, so I can connect nextcloudpi to WiFi :) Hope it help someone else

You probably now have NTP broken. See this question: https://unix.stackexchange.com/questions/788787/unable-to-get-ntp-systemd-timesyncd-working-over-wifi

marco-aa, 2025-01-17:
Hello all, Hello @3kimm0 

my workaround for Raspi 4B (maybe Pi 5 #2016 ) WLAN setup on 

"NextcloudPi v1.55.3 is up to date"
"v24.11.1 for Raspberry Pi 4 running Armbian Linux 6.6.63-current-bcm2711"

Check with iwconfig and/or networkctl if wlan0 module is loaded / installed

`sudo ifconfig`
lo        no wireless extensions.
end0      no wireless extensions.
wlan0     IEEE 802.11  [...]
          

`sudo networkctl`
IDX LINK TYPE     OPERATIONAL SETUP
  1 lo   loopback carrier     unmanaged
  2 end0 ether    routable    configured
  3 wlan0 wlan    [...]

if it's not listed/ installed 
```
sudo apt install  firmware-brcm80211
sudo reboot
```
and check again, if successfull

add netplan config
`sudo nano /etc/netplan/10-netplan-wlan0.yaml `
take care about yaml spacing/format

```
# Added by Armbian
#
# Reference: https://netplan.readthedocs.io/en/stable/netplan-yaml/
#
# Let systemd-networkd manage all Ethernet devices on this system, but be configured by Netplan.

network:
  version: 2
  renderer: networkd
  wifis:
    wlan0:
      dhcp4: true
      dhcp6: false
      access-points:
        "YourWLAN-SSID-Name":
          auth:
            key-management: "psk"
            password: "YourStrongSecretPW"

```

```
sudo chown root:root /etc/netplan/10-netplan-wlan0.yaml
sudo chmod 600 /etc/netplan/10-netplan-wlan0.yaml
```

check your new config
`sudo netplan try`
if ok
`sudo netplan apply`

should look like this
`sudo networkctl`
IDX LINK  TYPE     OPERATIONAL SETUP
  1 lo    loopback carrier     unmanaged
  2 end0  ether    routable    configured
  3 wlan0 wlan     routable    configured

check ip adress / dhcp

`ip a`

Based on the default metric 600 for wlan0 your lan access is preferred.
Disconnect lan cable or take down lan to check/use wlan0 - take care open session will be uselesse/closed
`sudo networkctl down end0 `

chepo92, 2025-04-11:
Expanding [marco-aa](https://github.com/marco-aa) solution, 

If `sudo netplan try`  throws `Cannot call openvswitch: ovsdb-server.service is not running.`
Do 
` sudo apt install openvswitch-switch-dpdk`

and continue with other steps

System: 
Pi4B, fresh new install NCPi 30.0.1 (and had to upgrade to 30.0.8)

SchroedersKater, 2025-08-02:
I tried all the suggestions here and in the other threads, but had no success. The WiFi connection is visible in my router, but there is not way to access it. I can't access it via SSH nor can I via browser. Only the ethernet connection is working.

Does someone have an idea what else I could try?

Here is some information from my system:

`                                                                         
 v25.5.1 for Raspberry Pi 4 Model B running Armbian Linux 6.12.32-current-bcm2711

 Packages:     Debian stable (bookworm)
 IPv4:         (LAN) 192.168.178.124 (WAN) 213.244.63.239 
 IPv6:         fd29:9ac6:a0e:0:665:bdeb:2f36:b8f6, fd29:9ac6:a0e:0:e65f:1ff:fed2:5a3a, 2a01:41e1:1d08:400:e65f:1ff:fed2:5a3a (WAN) 2a01:41e1:1d08:400:eff:ec8e:4b0b:75f5 

NextcloudPi v1.55.4 is up to date
 Performance:  

 Load:         20%           	Up time:       1 min	
 Memory usage: 19% of 1.79G  	
 CPU temp:     41°C           	 Usage of /:   39% of 14G    	
storage/:      29% of 253M   	

 Commands: 

 Monitoring : htop

Last login: Fri Aug  1 15:01:28 2025 from 192.168.178.200
thomas@nextcloudpi:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: end0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether e4:5f:01:d2:5a:3a brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.124/24 metric 100 brd 192.168.178.255 scope global dynamic end0
       valid_lft 863886sec preferred_lft 863886sec
    inet6 fd29:9ac6:a0e:0:665:bdeb:2f36:b8f6/64 scope global temporary dynamic 
       valid_lft 7086sec preferred_lft 3486sec
    inet6 fd29:9ac6:a0e:0:e65f:1ff:fed2:5a3a/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 7086sec preferred_lft 3486sec
    inet6 2a01:41e1:1d08:400:eff:ec8e:4b0b:75f5/64 scope global temporary dynamic 
       valid_lft 7086sec preferred_lft 3486sec
    inet6 2a01:41e1:1d08:400:e65f:1ff:fed2:5a3a/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 7086sec preferred_lft 3486sec
    inet6 fe80::e65f:1ff:fed2:5a3a/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether e4:5f:01:d2:5a:3b brd ff:ff:ff:ff:ff:ff
thomas@nextcloudpi:~$ ncp-config
PHP Warning:  include(/var/www/nextcloud/config/config.php): Failed to open stream: Permission denied in Command line code on line 1
PHP Warning:  include(): Failed opening '/var/www/nextcloud/config/config.php' for inclusion (include_path='.:/usr/share/php') in Command line code on line 1
PHP Warning:  Undefined variable $CONFIG in Command line code on line 1
PHP Warning:  Trying to access array offset on null in Command line code on line 1
Must be run as root. Try 'sudo ncp-config'
thomas@nextcloudpi:~$ sudo ncp-config
[sudo] password for thomas: 
thomas@nextcloudpi:~$ networkctl list
IDX LINK  TYPE     OPERATIONAL SETUP      
  1 lo    loopback carrier     unmanaged
  2 end0  ether    routable    configured 
  3 wlan0 wlan     no-carrier  configuring

3 links listed.
thomas@nextcloudpi:~$ sudo netplan apply
thomas@nextcloudpi:~$ networkctl list
IDX LINK  TYPE     OPERATIONAL SETUP      
  1 lo    loopback carrier     unmanaged
  2 end0  ether    routable    configured 
  3 wlan0 wlan     no-carrier  configuring

3 links listed.
thomas@nextcloudpi:~$ sudo networkctl status wlan0
● 3: wlan0                                                                     
                     Link File: /usr/lib/systemd/network/99-default.link
                  Network File: /run/systemd/network/10-netplan-wlan0.network
                         State: routable (configured)
                  Online state: online                                         
                          Type: wlan
                          Path: platform-fe300000.mmcnr
                        Driver: brcmfmac
              Hardware Address: e4:5f:01:d2:5a:3b (Raspberry Pi Trading Ltd)
                           MTU: 1500 (min: 68, max: 1500)
                         QDisc: pfifo_fast
  IPv6 Address Generation Mode: eui64
             WiFi access point: EuroWLAN5 (d4:24:dd:b4:3e:fd)
      Number of Queues (Tx/Rx): 1/1
                       Address: 192.168.178.224 (DHCP4 via 192.168.178.1)
                                2a01:41e1:1d08:400:e65f:1ff:fed2:5a3b
                                fd29:9ac6:a0e:0:e65f:1ff:fed2:5a3b
                                fe80::e65f:1ff:fed2:5a3b
                       Gateway: 192.168.178.1
                                fe80::d624:ddff:feb4:3efa
                           DNS: 192.168.178.1
                                fd29:9ac6:a0e:0:d624:ddff:feb4:3efa
                                2a01:41e1:1d08:400:d624:ddff:feb4:3efa
                           NTP: 192.168.178.1
                                2a01:41e1:1d08:400:d624:ddff:feb4:3efa
                                fd29:9ac6:a0e:0:d624:ddff:feb4:3efa
             Activation Policy: up
           Required For Online: yes
               DHCP4 Client ID: IAID:0x9539bff9/DUID
             DHCP6 Client IAID: 0x9539bff9
             DHCP6 Client DUID: DUID-EN/Vendor:0000ab117162a84e5fea6f01
thomas@nextcloudpi:~$ 
`