nextcloud/nextcloudpi #1977: Install script failing due to private DNS server requirement

gitarman94, 2024-09-15:
I may have found an issue with the installer script?  I've got protected DNS, and force all traffic on my network to go through my DNS server, all other traffic gets blocked (for security reasons).  The installer script kept failing saying it couldn't find the raw.githubusercontent.com website, but apt was able to install packages just fine prior to the attempted product install.  After looking at the code, I found the below...

cat <<'EOF' >/etc/systemd/resolved.conf
[Resolve]
DNS=8.8.8.8
FallbackDNS=8.8.4.4
EOF

It appears that you're manually pointing the lxc to google as a valid DNS server to resolve addresses.  In my case, this broke the installer as I'm blocking google DNS (for security/tracking purposes).  I was able to modify the file post failure to point at my DNS server, and then copy/paste the remainder of the script, but this code should probably default to the base system's dns server (or utilize the address that was prompted for during the setup wizard).

theCalcaholic, 2024-11-06:
Hi, thanks for reporting!
Where exactly did you find this in the code?

gitarman94, 2024-11-06:
Sorry, it looks like it was an installer script from Proxmox community-scripts, this issue can be closed and I’ll open a ticket with them.

closed 2024-11-06 by gitarman94

theCalcaholic, 2024-11-06:
Ah I see. For Proxmox I usually only test the official LXC image, but I have heard good things about the installer from tteck