nextcloud/nextcloudpi #1978: Problem upgrading to Nextcloud 30

silas229, 2024-09-16:
This error raised when doing migrations:

InvalidArgumentException: Index name "sh" for table "oc_social_3_stream_tag" collides with the constraint on table "oc_social_stream_tag"

<details>
<summary>Update log</summary>
<pre>
Running nc-update-nextcloud
Failed to reset failed state of unit ncp-encrypt.service: Unit ncp-encrypt.service not loaded.
Running as unit: ncp-update-nc.service
Current   Nextcloud version 29.0.3
Requested Nextcloud version 30.0.0
Selected  Nextcloud version 30.0.0
Download Nextcloud 30.0.0...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www/nextcloud-bkp_20240916_1726488993.tar.gz generated
Maintenance mode disabled
Storing backup at '/var/www/nextcloud-bkp_20240916_1726488993-29.0.3.tar.gz'...
Install Nextcloud 30.0.0...
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: announcementcenter
Disabled incompatible app: appointments
Disabled incompatible app: assistant
Disabled incompatible app: audioplayer
Disabled incompatible app: bookmarks
Disabled incompatible app: calendar
Disabled incompatible app: camerarawpreviews
Disabled incompatible app: cfg_share_links
Disabled incompatible app: checksum
Disabled incompatible app: contacts
Disabled incompatible app: cospend
Disabled incompatible app: deck
Disabled incompatible app: eidlogin
Disabled incompatible app: external
Disabled incompatible app: files_automatedtagging
Disabled incompatible app: files_linkeditor
Disabled incompatible app: files_snapshots
Disabled incompatible app: forms
Disabled incompatible app: gpoddersync
Disabled incompatible app: groupfolders
Disabled incompatible app: guests
Disabled incompatible app: impersonate
Disabled incompatible app: integration_deepl
Disabled incompatible app: integration_excalidraw
Disabled incompatible app: integration_github
Disabled incompatible app: integration_homeassistant
Disabled incompatible app: integration_mastodon
Disabled incompatible app: integration_openai
Disabled incompatible app: login_notes
Disabled incompatible app: mail
Disabled incompatible app: maps
Disabled incompatible app: memories
Disabled incompatible app: nextpod
Disabled incompatible app: notes
Disabled incompatible app: notify_push
Disabled incompatible app: ocs_api_viewer
Disabled incompatible app: oidc
Disabled incompatible app: pandoc
Disabled incompatible app: phonetrack
Disabled incompatible app: polls
Disabled incompatible app: qownnotesapi
Disabled incompatible app: quota_warning
Disabled incompatible app: recognize
Disabled incompatible app: repod
Disabled incompatible app: sendent
Disabled incompatible app: skyprint
Disabled incompatible app: sociallogin
Disabled incompatible app: spreed
Disabled incompatible app: tables
Disabled incompatible app: tasks
Disabled incompatible app: workflow_ocr
Disabled incompatible app: workflow_pdf_converter
Updating <federation> ...
Updated <federation> to 1.20.0
Updating <lookup_server_connector> ...
Updated <lookup_server_connector> to 1.18.0
Updating <oauth2> ...
Updated <oauth2> to 1.18.1
Updating <password_policy> ...
Updated <password_policy> to 2.0.0
Updating <photos> ...
Updated <photos> to 3.0.2
Updating <files> ...
Updated <files> to 2.2.0
Updating <activity> ...
Updated <activity> to 3.0.0
Updating <cloud_federation_api> ...
Updated <cloud_federation_api> to 1.13.0
Updating <dav> ...
Fix broken values of calendar objects

 Starting ...
    0/0 [>---------------------------]   0%
Clean up old calendar subscriptions from deleted users that were not cleaned-up

 Starting ...
  0/10 [>---------------------------]   0%
 Starting ...
 10/10 [============================] 100%
Updated <dav> to 1.31.1
Updating <files_sharing> ...
Updated <files_sharing> to 1.22.0
Updating <files_trashbin> ...
Updated <files_trashbin> to 1.20.1
Updating <files_versions> ...
Updated <files_versions> to 1.23.0
Updating <sharebymail> ...
Updated <sharebymail> to 1.20.0
Updating <workflowengine> ...
Updated <workflowengine> to 2.12.0
Updating <admin_audit> ...
Updated <admin_audit> to 1.20.0
Updating <comments> ...
Updated <comments> to 1.20.1
Updating <logreader> ...
Updated <logreader> to 3.0.0
Updating <nextcloud_announcements> ...
Updated <nextcloud_announcements> to 2.0.0
Updating <notifications> ...
Updated <notifications> to 3.0.0
Updating <systemtags> ...
Updated <systemtags> to 1.20.0
Updating <theming> ...
Updated <theming> to 2.5.0
Updating <bruteforcesettings> ...
Updated <bruteforcesettings> to 3.0.0
Updating <contactsinteraction> ...
Updated <contactsinteraction> to 1.11.0
Updating <dashboard> ...
Updated <dashboard> to 7.10.0
Updating <federatedfilesharing> ...
Updated <federatedfilesharing> to 1.20.0
Updating <files_downloadlimit> ...
Updated <files_downloadlimit> to 3.0.0
Updating <files_pdfviewer> ...
Updated <files_pdfviewer> to 3.0.0
Updating <files_reminders> ...
Updated <files_reminders> to 1.3.0
Updating <privacy> ...
Updated <privacy> to 2.0.0
Updating <provisioning_api> ...
Updated <provisioning_api> to 1.20.0
Updating <recommendations> ...
Updated <recommendations> to 3.0.0
Updating <related_resources> ...
Updated <related_resources> to 1.5.0
Updating <serverinfo> ...
Updated <serverinfo> to 2.0.0
Updating <settings> ...
Updated <settings> to 1.13.0
Updating <support> ...
Updated <support> to 2.0.0
Updating <survey_client> ...
Updated <survey_client> to 2.0.0
Updating <text> ...
Updated <text> to 4.1.0
Updating <twofactor_backupcodes> ...
Updated <twofactor_backupcodes> to 1.19.0
Updating <twofactor_nextcloud_notification> ...
Updated <twofactor_nextcloud_notification> to 4.0.0
Updating <twofactor_totp> ...
Updated <twofactor_totp> to 12.0.0-dev
Updating <user_status> ...
Updated <user_status> to 1.10.0
Updating <viewer> ...
Updated <viewer> to 3.0.0
Updating <weather_status> ...
Updated <weather_status> to 1.10.0
Add read permissions for existing announcements

 Starting ...
    0/0 [----->----------------------]   0%
Restore default tags that are missing

 Starting ...
 0/3 [>---------------------------]   0%
 Starting ...
 3/3 [============================] 100%
Purify and migrate collected mail addresses

 Starting ...
 0/1 [>---------------------------]   0%
 Starting ...
 1/1 [============================] 100%
Insert background jobs for all accounts

 Starting ...
 0/3 [>---------------------------]   0%
 Starting ...
 3/3 [============================] 100%
Add background jobs for existing users

 Starting ...
    0/0 [------------------>---------]   0%
InvalidArgumentException: Index name "sh" for table "oc_social_3_stream_tag" collides with the constraint on table "oc_social_stream_tag".
Update failed
Maintenance mode is kept active
Resetting log level
Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20240916_1726488993-29.0.3.tar.gz...
check free space...
extracting backup file /var/www/nextcloud-bkp_20240916_1726488993-29.0.3.tar.gz...
</pre>
</details>

Schniefel, 2024-09-17:
As I understand it, currently Nextcloud 29 is the maximum supported version by NextcloudPi

new-on-github, 2024-09-19:
> As I understand it, currently Nextcloud 29 is the maximum supported version by NextcloudPi

Think so, too. See the release info here:
https://github.com/nextcloud/nextcloudpi/releases/tag/v1.55.1

theCalcaholic, 2024-09-19:
> As I understand it, currently Nextcloud 29 is the maximum supported version by NextcloudPi

That's correct. The report might still be helpful once NC 30 support is being worked on

closed 2024-09-19 by theCalcaholic