nextcloud/nextcloudpi #1981: Permanent update notification despite system already updated…

dwprod38, 2024-09-21:
### ⚠️ This issue respects the following points: ⚠️

- [X] This is a **bug**, not a question or a configuration/webserver/proxy issue.
- [X] This issue is **not** already reported on [Github](https://github.com/nextcloud/server/issues?q=is%3Aopen+is%3Aissue+label%3Abug) OR [Nextcloud Community Forum](https://help.nextcloud.com/) _(I've searched it)_.
- [X] Nextcloud Server **is** up to date. See [Maintenance and Release Schedule](https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule) for supported versions.
- [X] I agree to follow Nextcloud's [Code of Conduct](https://nextcloud.com/contribute/code-of-conduct/).

### Bug description

Since update to 29.0.7, every day I’ve received random messages notification informing me “NexcloudPi , Nextcloud was updated to 29.0.7” despite that system was already updated since September 13th (see pictures red marks)

[image]

### Steps to reproduce

1. Nothing to do:  receiving automatic message notification every day…
2.
3.

### Expected behavior

When system is updated, the message notification of updating system must display one time after acknowledge it…

### Nextcloud Server version

29

### Operating system

Debian/Ubuntu

### PHP engine version

PHP 8.1

### Web server

Apache (supported)

### Database engine version

MySQL

### Is this bug present after an update or on a fresh install?

Updated from a MINOR version (ex. 28.0.1 to 28.0.2)

### Are you using the Nextcloud Server Encryption module?

None

### What user-backends are you using?

- [ ] Default user-backend _(database)_
- [ ] LDAP/ Active Directory
- [ ] SSO - SAML
- [X] Other

### Configuration report

```shell
root@nextcloudpiclone:/home/ncpadmin# sudo -u www-data php /var/www/nextcloud/occ config:list system
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "default_phone_region": "FR",
        "trusted_domains": {
            "0": "localhost",
            "7": "***REMOVED SENSITIVE VALUE***",
            "5": "nextcloudpiclone.local",
            "8": "nextcloudpiclone.lan",
            "3": "nextcloudpiclone",
            "11": "***REMOVED SENSITIVE VALUE***",
            "1": "192.168.1.38",
            "14": "nextcloudpiclone",
            "2": "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "29.0.7.1",
        "overwrite.cli.url": "https:\/\/nextcloudpiclone\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/mnt\/SSDdrive\/ncdata\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "maintenance_window_start": 1,
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "maintenance": false,
        "logfile": "\/mnt\/SSDdrive\/ncdata\/data\/nextcloud.log",
        "theme": "",
        "loglevel": 2
    }
}
```

### List of activated Apps

```shell
root@nextcloudpiclone:/home/ncpadmin# sudo -u www-data php /var/www/nextcloud/occ app:list 
Enabled:
  - activity: 2.21.1
  - calendar: 4.7.16
  - circles: 29.0.0-dev
  - cloud_federation_api: 1.12.0
  - comments: 1.19.0
  - contacts: 6.0.0
  - contactsinteraction: 1.10.0
  - dashboard: 7.9.0
  - dav: 1.30.1
  - federatedfilesharing: 1.19.0
  - federation: 1.19.0
  - files: 2.1.1
  - files_downloadlimit: 2.0.0
  - files_pdfviewer: 2.10.0
  - files_reminders: 1.2.0
  - files_sharing: 1.21.0
  - files_trashbin: 1.19.0
  - files_versions: 1.22.0
  - firstrunwizard: 2.18.0
  - logreader: 2.14.0
  - lookup_server_connector: 1.17.0
  - nextcloud_announcements: 1.18.0
  - notes: 4.11.0
  - notifications: 2.17.0
  - notify_push: 0.7.0
  - oauth2: 1.17.1
  - password_policy: 1.19.0
  - photos: 2.5.0
  - privacy: 1.13.0
  - provisioning_api: 1.19.0
  - recommendations: 2.1.0
  - related_resources: 1.4.0
  - serverinfo: 1.19.0
  - settings: 1.12.0
  - sharebymail: 1.19.0
  - support: 1.12.0
  - survey_client: 1.17.0
  - systemtags: 1.19.0
  - tasks: 0.16.1
  - text: 3.10.1
  - theming: 2.4.0
  - twofactor_backupcodes: 1.18.0
  - updatenotification: 1.19.1
  - user_status: 1.9.0
  - viewer: 2.3.0
  - weather_status: 1.9.0
  - workflowengine: 2.11.0
Disabled:
  - admin_audit: 1.19.0
  - bruteforcesettings: 2.9.0
  - encryption: 2.17.0
  - files_external: 1.21.0
  - nextcloudpi: 0.0.2
  - previewgenerator: 99.99.99
  - suspicious_login: 7.0.0
  - twofactor_totp: 11.0.0-dev
  - user_ldap: 1.20.0
```

dwprod38, 2024-09-24:
And same issue with second nextcloudpi server with version 28.0.10...

[image]

theCalcaholic, 2024-09-24:
Fixed in v1.55.2

closed 2024-09-24 by theCalcaholic