nextcloud/nextcloudpi #1988: reset password for ncp 4443 login - lost it

rougerouter, 2024-09-28:
Thanks for reporting issues back to NextcloudPi! 

Here you can file bugs and feature requests. **Do NOT ask questions**, this is not a support forum.

---
DO NOT ASK QUESTIONS, USE THE FORUMS
---

NCP and its extras are explained in the wiki, also make sure to read the articles in ownyourbits explaining NextcloudPi extras before asking. https://ownyourbits.com/category/nextcloudpi/

We use GitHub issues only to discuss about NextcloudPi bugs and new features.

NCP and its extras are explained in the [wiki](https://github.com/nextcloud/nextcloudpi/wiki), also make sure to read the [articles in ownyourbits](https://ownyourbits.com/category/nextcloudpi/) explaining NextcloudPi extras before asking.

For any other question or comment, please use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm), and make sure to tag
the thread as `ncp`.

Note that quite probably your question has an answer in the [FAQ](https://github.com/nextcloud/nextcloudpi/wiki/FAQ).

Thank you!

rougerouter, 2024-09-28:
I am using the nextcloudpi version, not an appliance like docker or any other.

FadeFx, 2024-10-15:
Hi, as this did not receive an answer for long, i would like to help you there...

first of all, this is not a support forum, you should have posted your question on [help.nextcloud.com](url) but i think (i am not totally sure thou, the NCP login here uses the password of the nextcloud NCP user. so if yu have an admin account in nextcloud besides the NCP user, you should be able to set a new password for that user and log in to the ncp-panel with the new password. But as i say, i am not completely sure about this.

rougerouter, 2024-10-15:
> [Tracking image][https://github.com/notifications/beacon/ABASMDEXRE7JNILPA7NWVDLZ3VLYNA5CNFSM6AAAAABPAVFTF6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUP5V3NU.gif]

Thanks for the alert and no, creating a new user admin pswd for ncp does not let you login to the nc panel.

FadeFx, 2024-10-15:
Do you have console access?
Then you can use sudo ncp-config
there in config/nc-password is an option to set a new password for the panel.

theCalcaholic, 2024-11-06:
Hi @rougerouter, @FadeFx is right, your question is better suited for the [forum](https://help.nextcloud.com).

But let me give you a wrap-up of your options:

You need either

- ssh/console access to your server or
- you need to find a way to get that kind of access, for example by searching in the [Armbian forums](https://forum.armbian.com) for a solution to get access to your instance after losing the password (in case you're running one of the NextcloudPi images for single board computers like the Raspberry Pi). I'm sure you will find help there

After you gained access to the terminal, you can follow @FadeFx' recommendation to set the ncp admin password from `sudo ncp-config`

closed 2024-11-06 by theCalcaholic

rougerouter, 2024-11-07:
Thank you very much!