nextcloud/nextcloudpi #1990: Installation fails on Proxmox Container because of installing new systemd-resolve

Flappiee, 2024-10-09:
Hi,

I created a new Proxmox container en tried to install NextCloudPi.
The installation fails because the script installs a new systemd-resolve with a new /etc/resolv.conf with a wrong DNS server.

First of all, after creating the container, the /etc/resolv.conf looks like this:
`# --- BEGIN PVE ---
search lan.snowice.nl
nameserver 1.1.1.1
nameserver 9.9.9.9
# --- END PVE ---`

When the script installs the new systemd-resolve the file looks like this:
`
nameserver 127.0.0.53
options edns0 trust-ad
search .
`

When this wrong DNS server has been set, the script can't continue because it can't reach http://deb.debian.org anymore.

bub4, 2024-10-13:
Maybe you want to try ttecks scirpt?

https://tteck.github.io/Proxmox/

just search for NextCloudPi and the wonder will happen :)

theCalcaholic, 2024-10-14:
Hi, for Proxmox you can just use the official NextcloudPi LXC image from the releases or ttecks install script.

However, this will hopefully be fixed in the next release

theCalcaholic, 2024-11-06:
I'm gonna close this as duplicate for #1926 for now (which will be fixed soon)

closed 2024-11-06 by theCalcaholic