nextcloud/nextcloudpi #1992: Updating fails with: Unit ncp-update-nc.service was already loaded or has a fragment file

janvlug, 2024-10-10:
When trying to run nc-update-nextcloud from the web interface, I get:

```
[ nc-update-nextcloud ] (Thu Oct 10 15:35:20 BST 2024)
Failed to start transient service unit: Unit ncp-update-nc.service was already loaded or has a fragment file.
```

[image]

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.55.2
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.3.img
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/nextcloudpi_data/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      2.7T/5.3T
rootfs usage         7.4G/59G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    29.0.7.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               31days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "vlug.info",
            "11": "redacted",
            "1": "192.168.178.15",
            "2": "redacted.com",
            "14": "nextcloudpi"
[... 82 lines cut ...]
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"Zwd7Qr5B2IfMLiXbU9LRWwAAhgU","level":2,"time":"2024-10-10T06:59:25+00:00","remoteAddr":"86.85.119.40","user":"--","app":"core","method":"OPTIONS","url":"/remote.php/webdav","message":"Login failed: 'janjr' (Remote IP: '86.85.119.40')","userAgent":"gnome-online-accounts/3.50.2 libsoup/3.4.4","version":"29.0.7.1","data":{"app":"core"}}
{"reqId":"Zwd7pkF004GB1JTmWQEqBQAAAAs","level":3,"time":"2024-10-10T07:00:54+00:00","remoteAddr":"86.85.119.40","user":"janjr","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/user_status/api/v1/user_status?format=json","message":"unlink(/media/nextcloudpi_data/ncdata/data/appdata_ocrrxqnhospw/richdocuments/remoteData/discovery): No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#348","userAgent":"Mozilla/5.0 (Linux) mirall/3.13.2git (Nextcloud, fedora-6.10.6-200.fc40.x86_64 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"29.0.7.1","data":{"app":"PHP"}}
{"reqId":"Zwd7pjkzU1KAwEmsJrd2mQAAAFY","level":3,"time":"2024-10-10T07:00:54+00:00","remoteAddr":"86.85.119.40","user":"janjr","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/notifications/api/v2/notifications?format=json","message":"unlink(/media/nextcloudpi_data/ncdata/data/appdata_ocrrxqnhospw/richdocuments/remoteData/capabilities): No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#348","userAgent":"Mozilla/5.0 (Linux) mirall/3.13.2git (Nextcloud, fedora-6.10.6-200.fc40.x86_64 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"29.0.7.1","data":{"app":"PHP"}}
{"reqId":"Zwd74jkzU1KAwEmsJrd2oAAAAFI","level":3,"time":"2024-10-10T07:01:54+00:00","remoteAddr":"86.85.119.40","user":"janjr","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/user_status/api/v1/user_status?format=json","message":"unlink(/media/nextcloudpi_data/ncdata/data/appdata_ocrrxqnhospw/richdocuments/remoteData/discovery): No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#348","userAgent":"Mozilla/5.0 (Linux) mirall/3.13.2git (Nextcloud, fedora-6.10.6-200.fc40.x86_64 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"29.0.7.1","data":{"app":"PHP"}}
{"reqId":"Zwd74jkzU1KAwEmsJrd2oAAAAFI","level":3,"time":"2024-10-10T07:01:54+00:00","remoteAddr":"86.85.119.40","user":"janjr","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/user_status/api/v1/user_status?format=json","message":"unlink(/media/nextcloudpi_data/ncdata/data/appdata_ocrrxqnhospw/richdocuments/remoteData/capabilities): No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#348","userAgent":"Mozilla/5.0 (Linux) mirall/3.13.2git (Nextcloud, fedora-6.10.6-200.fc40.x86_64 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"29.0.7.1","data":{"app":"PHP"}}
<redacted>
{"reqId":"Zwe-Abbyv0oLXdIlKJk5WQAAABI","level":3,"time":"2024-10-10T11:44:01+00:00","remoteAddr":"178.79.139.171","user":"--","app":"PHP","method":"GET","url":"/nmaplowercheck1728560641","message":"file_get_contents(/media/nextcloudpi_data/ncdata/data/appdata_ocrrxqnhospw/richdocuments/remoteData/capabilities): Failed to open stream: No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#331","userAgent":"curl/7.54.0","version":"29.0.7.1","data":{"app":"PHP"}}
{"reqId":"Zwe-Abbyv0oLXdIlKJk5WQAAABI","level":3,"time":"2024-10-10T11:44:01+00:00","remoteAddr":"178.79.139.171","user":"--","app":"index","method":"GET","url":"/nmaplowercheck1728560641","message":"Exception thrown: OCP\\Files\\GenericFileException","userAgent":"curl/7.54.0","version":"29.0.7.1","exception":{"Exception":"OCP\\Files\\GenericFileException","Message":"","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Files/SimpleFS/SimpleFile.php","line":74,"function":"getContent","class":"OC\\Files\\Node\\File","type":"->"},{"file":"/var/www/nextcloud/apps/richdocuments/lib/Service/CachedRequestService.php","line":47,"function":"getContent","class":"OC\\Files\\SimpleFS\\SimpleFile","type":"->"},{"file":"/var/www/nextcloud/apps/richdocuments/lib/Service/CapabilitiesService.php","line":147,"function":"get","class":"OCA\\Richdocuments\\Service\\CachedRequestService","type":"->"},{"file":"/var/www/nextcloud/apps/richdocuments/lib/Service/CapabilitiesService.php","line":63,"function":"getParsedCapabilities","class":"OCA\\Richdocuments\\Service\\CapabilitiesService","type":"->"},{"file":"/var/www/nextcloud/apps/richdocuments/lib/Service/CapabilitiesService.php","line":91,"function":"getCapabilities","class":"OCA\\Richdocuments\\Service\\CapabilitiesService","type":"->"},{"file":"/var/www/nextcloud/apps/richdocuments/lib/Capabilities.php","line":133,"function":"hasDrawSupport","class":"OCA\\Richdocuments\\Service\\CapabilitiesService","type":"->"},{"file":"/var/www/nextcloud/lib/private/CapabilitiesManager.php","line":74,"function":"getCapabilities","class":"OCA\\Richdocuments\\Capabilities","type":"->"},{"file":"/var/www/nextcloud/lib/private/Template/JSConfigHelper.php","line":157,"function":"getCapabilities","class":"OC\\CapabilitiesManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/TemplateLayout.php","line":266,"function":"getConfig","class":"OC\\Template\\JSConfigHelper","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Template.php","line":145,"function":"__construct","class":"OC\\TemplateLayout","type":"->"},{"file":"/var/www/nextcloud/lib/private/Template/Base.php","line":132,"function":"fetchPage","class":"OC_Template","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":859,"function":"printPage","class":"OC\\Template\\Base","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":1189,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/index.php","line":47,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Files/Node/File.php","Line":56,"message":"","exception":{},"CustomMessage":"Exception thrown: OCP\\Files\\GenericFileException"}}
<redacted>
```
</details>

geoW-n, 2024-10-11:
You can fix it on cli:

- `sudo systemctl status ncp-update-nc.service`
- `sudo systemctl restart ncp-update-nc.service`
- do the nc-update
It should be done (you may check again).

janvlug, 2024-10-12:
Thanks. The instructions above worked.

For interested people, first I did this on the command line:
<details>

```
jan@nextcloudpi:~ $ sudo systemctl status ncp-update-nc.service
[sudo] password for jan: 
× ncp-update-nc.service - /usr/bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log
     Loaded: loaded (/run/systemd/transient/ncp-update-nc.service; transient)
  Transient: yes
     Active: failed (Result: exit-code) since Sun 2024-10-06 20:20:30 BST; 5 days ago
   Duration: 4.703s
   Main PID: 2661529 (code=exited, status=1/FAILURE)
        CPU: 4.453s

Oct 06 20:20:29 nextcloudpi bash[2661531]:                                  Dload  Upload   Total   Spent    Left  Speed
Oct 06 20:20:30 nextcloudpi bash[2661531]: [237B blob data]
Oct 06 20:20:30 nextcloudpi bash[2661531]: Nextcloud version 29.0.7 is already installed. Nothing to do.
Oct 06 20:20:30 nextcloudpi systemd[1]: ncp-update-nc.service: Main process exited, code=exited, status=1/FAILURE
Oct 06 20:20:30 nextcloudpi systemd[1]: ncp-update-nc.service: Failed with result 'exit-code'.
Oct 06 20:20:30 nextcloudpi systemd[1]: ncp-update-nc.service: Consumed 4.453s CPU time.
Oct 09 06:18:11 nextcloudpi systemd[1]: /run/systemd/transient/ncp-update-nc.service:7: Ignoring unknown escape sequences: "set -o pipefail\; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log"
Oct 09 06:18:13 nextcloudpi systemd[1]: /run/systemd/transient/ncp-update-nc.service:7: Ignoring unknown escape sequences: "set -o pipefail\; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log"
Oct 09 06:18:15 nextcloudpi systemd[1]: /run/systemd/transient/ncp-update-nc.service:7: Ignoring unknown escape sequences: "set -o pipefail\; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log"
Oct 09 06:18:18 nextcloudpi systemd[1]: /run/systemd/transient/ncp-update-nc.service:7: Ignoring unknown escape sequences: "set -o pipefail\; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log"
jan@nextcloudpi:~ $ sudo systemctl restart ncp-update-nc.service
jan@nextcloudpi:~ $ sudo systemctl status ncp-update-nc.service
Unit ncp-update-nc.service could not be found.
```

</details>

After that, I ran ncp_update_nextcloud in the web interface, and got this:

<details>

```
[ nc-update-nextcloud ] (Sat Oct 12 16:43:39 BST 2024)
Running as unit: ncp-update-nc.service
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed

Current Nextcloud version 29.0.7
Requested Nextcloud version latest
Selected Nextcloud version 29.0.8
INFO: You have requested an update to 'latest', but a direct update to 'latest' cannot be performed, so the latest available version that can be updated to (29.0.8) has been selected automatically.
Download Nextcloud 29.0.8...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www/nextcloud-bkp_20241012_1728747864.tar.gz generated
Maintenance mode disabled
Storing backup at '/var/www/nextcloud-bkp_20241012_1728747864-29.0.7.tar.gz'...
Install Nextcloud 29.0.8...
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: announcementcenter
Disabled incompatible app: calendar
Disabled incompatible app: contacts
Disabled incompatible app: deck
Disabled incompatible app: mail
Disabled incompatible app: maps
Disabled incompatible app: notes
Disabled incompatible app: notify_push
Disabled incompatible app: richdocuments
Disabled incompatible app: richdocumentscode_arm64
Disabled incompatible app: spreed
Disabled incompatible app: tasks
Add read permissions for existing announcements

Starting ...
0/0 [>---------------------------] 0%
Add read permissions for existing announcements

<Here I removed part of the output for brevity>

Clean up...
Nextcloud update finished successfully.
```

</details>

I'm not sure if there is still something that has to be fixed, or if this issue will not appear for others and can be closed. So I leave the issue open for now.

sebeweiss, 2024-10-19:
You can also reboot your server. I faced the same problem.

frankgroneberg, 2024-11-01:
Hello,
I have the same error message. However, entering the commands and rebooting did not lead to success. The error message remains :-(
Greetings Frank

geoW-n, 2024-11-01:
Sorry to hear this, to clear things it should go as [janvlug]( https://github.com/nextcloud/nextcloudpi/issues/1992#issuecomment-2408610874) shows in details.
Dealing with systemctl is the alternative method to reboot.
Checking service again it should not be there, then run update routine.

frankgroneberg, 2024-11-01:
Hi,
The following messages appear:

```
pi@nextcloudpi:~ $ systemctl status ncp-update-nc.service
× ncp-update-nc.service - /usr/bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log
     Loaded: loaded (/run/systemd/transient/ncp-update-nc.service; transient)
  Transient: yes
     Active: failed (Result: exit-code) since Fri 2024-11-01 19:02:16 CET; 14min ago
   Duration: 3.323s
    Process: 9400 ExecStart=/usr/bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log (code=exited, status=1/FAILURE)
   Main PID: 9400 (code=exited, status=1/FAILURE)
        CPU: 3.289s

Nov 01 19:02:15 nextcloudpi sudo[9431]:     root : PWD=/var/www ; USER=www-data ; COMMAND=/usr/bin/php -r 'include("/var/www/nextcloud/config/config.php"); echo($CONFIG["datadirectory"]);'
Nov 01 19:02:15 nextcloudpi sudo[9431]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Nov 01 19:02:15 nextcloudpi sudo[9431]: pam_unix(sudo:session): session closed for user www-data
Nov 01 19:02:15 nextcloudpi sudo[9434]:     root : PWD=/var/www ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ status
Nov 01 19:02:15 nextcloudpi sudo[9434]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Nov 01 19:02:16 nextcloudpi sudo[9434]: pam_unix(sudo:session): session closed for user www-data
Nov 01 19:02:16 nextcloudpi bash[9402]: Nextcloud backup directory found. Interrupted or already running installation?
Nov 01 19:02:16 nextcloudpi systemd[1]: ncp-update-nc.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 19:02:16 nextcloudpi systemd[1]: ncp-update-nc.service: Failed with result 'exit-code'.
Nov 01 19:02:16 nextcloudpi systemd[1]: ncp-update-nc.service: Consumed 3.289s CPU time.
pi@nextcloudpi:~ $ sudo systemctl restart ncp-update-nc.service
pi@nextcloudpi:~ $ systemctl status ncp-update-nc.service
× ncp-update-nc.service - /usr/bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log
     Loaded: loaded (/run/systemd/transient/ncp-update-nc.service; transient)
  Transient: yes
     Active: failed (Result: exit-code) since Fri 2024-11-01 19:16:49 CET; 10s ago
   Duration: 3.781s
    Process: 10379 ExecStart=/usr/bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh '0' |& tee /var/log/ncp-update-nc.log (code=exited, status=1/FAILURE)
   Main PID: 10379 (code=exited, status=1/FAILURE)
        CPU: 3.743s

Nov 01 19:16:47 nextcloudpi sudo[10408]:     root : PWD=/var/www ; USER=www-data ; COMMAND=/usr/bin/php -r 'include("/var/www/nextcloud/config/config.php"); echo($CONFIG["datadirectory"]);'
Nov 01 19:16:47 nextcloudpi sudo[10408]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Nov 01 19:16:47 nextcloudpi sudo[10408]: pam_unix(sudo:session): session closed for user www-data
Nov 01 19:16:47 nextcloudpi sudo[10411]:     root : PWD=/var/www ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ status
Nov 01 19:16:47 nextcloudpi sudo[10411]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Nov 01 19:16:49 nextcloudpi sudo[10411]: pam_unix(sudo:session): session closed for user www-data
Nov 01 19:16:49 nextcloudpi bash[10381]: Nextcloud backup directory found. Interrupted or already running installation?
Nov 01 19:16:49 nextcloudpi systemd[1]: ncp-update-nc.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 19:16:49 nextcloudpi systemd[1]: ncp-update-nc.service: Failed with result 'exit-code'.
Nov 01 19:16:49 nextcloudpi systemd[1]: ncp-update-nc.service: Consumed 3.743s CPU time.
pi@nextcloudpi:~ $ 
``` 
I am at a loss

geoW-n, 2024-11-01:
Run it with sudo
To get rid of the error:
`sudo systemctl restart ncp-update-nc.service` or reboot
To check:
`sudo systemctl status ncp-update-nc.service`
If its vanished do the update.

geoW-n, 2024-11-01:
Take a look:
`Nov 01 19:16:49 nextcloudpi bash[10381]: Nextcloud backup directory found. Interrupted or already running installation?`

I must admit that my solution does not sshow up as described.
```
pi@nextcloudpi:~ $ sudo systemctl status ncp-update-nc.service
× ncp-update-nc.service - /bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh 'latest' |& >
     Loaded: loaded (/run/systemd/transient/ncp-update-nc.service; transient)
  Transient: yes
     Active: failed (Result: exit-code) since Fri 2024-11-01 06:25:54 CET; 13h ago
   Duration: 4.022s
    Process: 74713 ExecStart=/bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh 'latest' >
   Main PID: 74713 (code=exited, status=1/FAILURE)
        CPU: 4.017s

Nov 01 06:25:53 nextcloudpi sudo[74762]:     root : PWD=/var/www ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcl>
Nov 01 06:25:53 nextcloudpi sudo[74762]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Nov 01 06:25:54 nextcloudpi sudo[74762]: pam_unix(sudo:session): session closed for user www-data
Nov 01 06:25:54 nextcloudpi bash[74716]:   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Nov 01 06:25:54 nextcloudpi bash[74716]:                                  Dload  Upload   Total   Spent    Left  Speed
Nov 01 06:25:54 nextcloudpi bash[74716]: [237B blob data]
Nov 01 06:25:54 nextcloudpi bash[74716]: Nextcloud version 29.0.8 is already installed. Nothing to do.
Nov 01 06:25:54 nextcloudpi systemd[1]: ncp-update-nc.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 06:25:54 nextcloudpi systemd[1]: ncp-update-nc.service: Failed with result 'exit-code'.
Nov 01 06:25:54 nextcloudpi systemd[1]: ncp-update-nc.service: Consumed 4.017s CPU time.
pi@nextcloudpi:~ $ sudo systemctl restart ncp-update-nc.service
pi@nextcloudpi:~ $ sudo systemctl status ncp-update-nc.service
× ncp-update-nc.service - /bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh 'latest' |& >
     Loaded: loaded (/run/systemd/transient/ncp-update-nc.service; transient)
  Transient: yes
     Active: failed (Result: exit-code) since Fri 2024-11-01 19:51:27 CET; 1s ago
   Duration: 4.238s
    Process: 90879 ExecStart=/bin/bash -c set -o pipefail; DBG='' /usr/local/bin/ncp-update-nc.d/update-nc.sh 'latest' >
   Main PID: 90879 (code=exited, status=1/FAILURE)
        CPU: 3.921s

Nov 01 19:51:26 nextcloudpi sudo[90921]:     root : PWD=/var/www ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcl>
Nov 01 19:51:26 nextcloudpi sudo[90921]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Nov 01 19:51:27 nextcloudpi sudo[90921]: pam_unix(sudo:session): session closed for user www-data
Nov 01 19:51:27 nextcloudpi bash[90881]:   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Nov 01 19:51:27 nextcloudpi bash[90881]:                                  Dload  Upload   Total   Spent    Left  Speed
Nov 01 19:51:27 nextcloudpi bash[90881]: [237B blob data]
Nov 01 19:51:27 nextcloudpi bash[90881]: Nextcloud version 29.0.8 is already installed. Nothing to do.
Nov 01 19:51:27 nextcloudpi systemd[1]: ncp-update-nc.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 19:51:27 nextcloudpi systemd[1]: ncp-update-nc.service: Failed with result 'exit-code'.
Nov 01 19:51:27 nextcloudpi systemd[1]: ncp-update-nc.service: Consumed 3.921s CPU time.
pi@nextcloudpi:~ $
```

frankgroneberg, 2024-11-01:
Nothing changes with Sudo.
Where do I look for the backup? And what do I do with it?

geoW-n, 2024-11-01:
> Where do I look for the backup? And what do I do with it?

Depends on your setup. Take a look in the logs :4443 WebU you can download them, search for *tar,
mv it, reboot and try to update.

frankgroneberg, 2024-11-01:
Hi,
I have found the backup directory and moved it to another directory. Now the update starts again...hurray.
Hope it goes through.

Penguinjumper, 2024-11-02:
same issue here

frankgroneberg, 2024-11-02:
Hello,
I have deleted 
/var/www/nextcloud-old 
After that 
sudo systemctl restart ncp-update-nc.service
and started the update via web-config.
Everything worked.

theCalcaholic, 2024-11-08:
Fixed in v1.55.3

PR #2000 (2024-11-06): NC 30 + fixes
Fixes #1992 
Fixes #1926
commit a3ccf157 (2024-11-06): Merge pull request #2000 from nextcloud/fix/networking-raspbian

- Add NC 30 support
- Fixes #1992
- Fixes #1926
closed 2024-11-08 by theCalcaholic