nextcloud/nextcloudpi #1994: Activation not Possible

AndreHerling, 2024-10-15:
Hi,

I have a Raspberry PI 5 with 8GB Ram. As router I have a Fritz!Box.

I downloaded the image for v1.55.2 and flashed it to a micro SD with Raspberry PI Imager. Then, according to the manual (https://help.nextcloud.com/t/how-to-access-nextcloudpi-first-time-activation/126309), I was supposed to access https://nextcloudpi or https://nextcloudpi.local. Both sites were not accessible. I also tried to access the IP, but it just doesn't work and looking to the comments on https://help.nextcloud.com/t/how-to-access-nextcloudpi-first-time-activation/126309 I'm not the only one. 

The  Raspberry PI 5 is seen in the network and I can ping it.

Greetings

André

paulmoench, 2024-10-15:
Hey,
thanks for creating the topic. I have the same issue and cannot access the activation site. When I try it either with the link or the IP address, it says Error 403 - no authorization for accessing this page. 
For setup I am using a Raspberry Pi 4 4gb and a MicroSD with a capacity of 32gb. My router is a Fritz!Box as well. Before, I was able to flash Raspbian and other Linux images onto my SD card and use them with the Pi. For flashing I have tried Balena Etcher and the Raspberry Imager. I also tried different versions of the Nextcloudpi image: v1.55.2, v1.55.1, v1.54.0. The error occurs with either of these images. 
When connecting the Pi via HDMI to a screen, the booting process seems to work fine, although I am not an expert. I can also find the pi in the Interface of my router and find it via SSH even though access over SSH is not possible because I haven't enabled it. 
In another (German) thread, I read, that the WebUI, for some reason was not enabled and therefore producing this problem. Unfortunately I cannot access the config to activate the WebUI in the TUI, because I have no access to the random password that is generated and usually displayed during the activation process. pi/raspberry and ncp/ownyourbits do not work.
Does anyone have a solution? Thanks in advance :)

FadeFx, 2024-10-19:
Did you guys try activating it via CLI?
When connecting via SSH you can enter
sudo ncp-config
There go to config/nc-webui and set it to enabled if it is not.
You could also even enable nextcloud using the CLI instead of the webui

gwilherm, 2024-10-28:
Hello, I have the same issue with RaspberryPi 4 using NextcloudPi_RaspberryPi4_v1.55.2.zip (md5sum checked).
https://nextcloudpi.local gives the raw content of PHP code : 

```php
<?php
/**
 * @copyright Copyright (c) 2016, ownCloud, Inc.
 *
 * @author Christoph Wurst <christoph@winzerhof-wurst.at>
 * @author Joas Schilling <coding@schilljs.com>
 * ...
```
https://192.168.1.36 brings me to the activation page, but it is not working : 

> NextCloudPi not yet initialized, trying again in a few seconds ...

I don't have microHDMI cable, so as says FadeFx I tried SSH. But SSH is not activated (nmap says port 22 not opened, btw 4443 port is closed aswell)
I tried to touch a /boot/firmware/ssh file (raspbian thing), no SSH.
I tried to touch a /boot/ssh file (old raspbian thing ?), still no SSH.

For the long story : 
I was running a NextcloudPi 1.55.1 / bullseye.
Update to 1.55.2 was not working so I thought maybe I had to do the bookworm upgrade.
So I ran ncp-dist-upgrade, and after more than one hour of package installation it (guess what) failed.
Then ncp commands mostly broken (no php interpreter). So I tried the fresh install. Here I am...

AndreHerling, 2024-10-29:
I installed the v.1.55.0 image and then updated it in the WebUI to v.1.55.2. This worked well.

gwilherm, 2024-10-29:
Thank you @AndreHerling your workaround worked well for me too

Bingo2023, 2024-11-06:
local login (w/ keyboard connected to Pi5) also not working, SSH can also not be activated, very annoying.

Will try with older image version <v1.55.2

FadeFx, 2024-11-06:
> local login (w/ keyboard connected to Pi5) also not working, SSH can also not be activated, very annoying.
> 
> Will try with older image version <v1.55.2

Did you try putting the SSH file in root partition of the SD card?

AndreHerling, 2024-11-06:
It is intended to activate SSH from the WebUI.

FadeFx, 2024-11-06:
I suspect 1.55.2 to have issues with root access, that is why several things do not work

theCalcaholic, 2024-11-06:
Sorry to hear about your issues. Unfortunately, the information isn't really conclusive for me yet and I don't know what might be causing these issues.
Most likely, the v1.55.2 image for the RPi 5 (and 4?) is broken in some way, despite it having worked in the automated tests.

I'll release a new version soon which hopefully works better.

A few notes though:

1. Depending on your network configuration, it might not work to access nextcloudpi via https://nextcloudpi.local or https://nextcloudpi. The safest bet is to access the activation page directly via the IP address (with https:// but without any port).
2. Raspbian related configurations or workaround won't work with the NextcloudPi images, because they are now based on Armbian. You can usually do the same with armbian, but it will work differently - please refer to the Armbian [documentation](https://docs.armbian.com/) and [help forum](https://forum.armbian.com).

FadeFx, 2024-11-06:
When did the switch to armbian happen?
I actually expect the 1.55.2 version to fail updating nextcloud as well, as long as the mysql root user password is not empty, but there has not yet been an update yet...

FadeFx, 2024-11-06:
To be more precise, I think there is an issue to access /root/.my.cnf

FadeFx, 2024-11-07:
BTW according to armbian doku SSH should be enabled by default with user root and password 1234 which you should be prompted to change on first login and create a proper user as well.

theCalcaholic, 2024-11-07:
> BTW according to armbian doku SSH should be enabled by default with user root and password 1234 which you should be prompted to change on first login and create a proper user as well.

Default root credentials as well as default SSH access are disabled in NCP, because that would pose a big security risk because of the activation mechanism that NCP uses.

theCalcaholic, 2024-11-07:
> When did the switch to armbian happen?
> I actually expect the 1.55.2 version to fail updating nextcloud as well, as long as the mysql root user password is not empty, but there has not yet been an update yet...

NCP is using Armbian for all SBCs since v1.54.0

theCalcaholic, 2024-11-07:
> To be more precise, I think there is an issue to access /root/.my.cnf

What makes you believe that? 🤔

FadeFx, 2024-11-07:
> > To be more precise, I think there is an issue to access /root/.my.cnf
> 
> What makes you believe that? 🤔

As for me I can only update nextcloud when mysql(mariadb) root password is empty. Otherwise I get an error that it can not login to mysql

theCalcaholic, 2024-11-08:
> > > To be more precise, I think there is an issue to access /root/.my.cnf
> > 
> > 
> > What makes you believe that? 🤔
> 
> As for me I can only update nextcloud when mysql(mariadb) root password is empty. Otherwise I get an error that it can not login to mysql

Would you mind creating another bug report for that? It's likely unrelated to this one and I would need the output of the failing commands to understand the cause for this.

theCalcaholic, 2024-11-08:
Now that v1.55.3 is released, please try and report if these issues persist for your :)

joehai, 2024-11-09:
Hi, 
I use BalenaEther (V1.5.113; MacOS) to copy the image (for PI4) on a 32GB SD card and I'm facing the same behaviour with the latest release 1.55.3:
PI boots, can be located in the network (ping works) but cannot be reached under https://192.168.178.25/ nor https://nextcloudpi/ or https://nextcloudpi.local/. nslookup provides nextcloudpi.fritz.box as name (which does also not work). 
The Raspberry is a PI 4 B 2GB, it works well with other images/installations, so I do not think it is a (obvious) hardware issue. But it has a Hifiberry-Soundcard on top. 
It is directly connected via ethernet to a FritzBox. 

Same behaviour with 1.55.0 image.

geoW-n, 2024-11-12:
> 32GB SD card and I'm facing the same behaviour with the latest release 1.55.3:

Did a quick install with release 1.55.3 on rpi4 sd-card 32GB, wrote image on Win10 laptop with Raspberrry Pi Imager v.1.9.0.
Up and running as expected in minutes.

F4HTB, 2024-11-12:
Same here. 1.55.3 on RPI5, can't connect on login, ssh activated but can't connect, activation interface dont get response.....

theCalcaholic, 2024-11-12:
> Hi,
> I use BalenaEther (V1.5.113; MacOS) to copy the image (for PI4) on a 32GB SD card and I'm facing the same behaviour with the latest release 1.55.3:
> PI boots, can be located in the network (ping [...]

Do you have any way to attach a display and check if there are any meaninful logs during boot?

theCalcaholic, 2024-11-12:
@geoW-n Thanks for testing!

geoW-n, 2024-11-12:
> Did a quick install with release 1.55.3 on rpi4 sd-card 32GB, wrote image on Win10 laptop with Raspberrry Pi Imager v.1.9.0.
> Up and running as expected in minutes.

Did a quick test with raspi 5. It works as expected.

- Keep iut simple, rpi5, sd card, ethernet, power
- Don't use presets. Write only the img to the sd-card.
- In a few minutes it is up and running, check ip address in router
- Take a web-browser with javascript enabled, https://_ip-address_ 
- Activation page. print it - activate it, every reloading before activation changes credentials
- redirection to webui login https://ip-address:4443
- first thing I recommend is activate ssh
**Now you can get comfortable with your server and have fun!!**

Belaial, 2025-01-29:
I just downloaded NextcloudPi_LXC_x86_v1.55.3.tar.gz (checked MD5 also, and it's ok)

Loaded this up in my Proxmox as an LXC, it boots up just fine, I can get to the activation page but once I hit activate it just "NextCloudPi not yet initialized, trying again in a few seconds ..."

Any tips? I have tried to re-create the LXC but same thing happens again.
(I also waited and waited and waited 😄 , did a restart of LXC also but nothing helps)

my-letterbox-3141, 2025-02-11:
Just ran into the same problem with a new pi5 & v1.55.3 [power, ethernet, 128GB sd-card, pi5 -> pingable but not reachable via browser].

neoscaler, 2025-03-13:
Still happening with latest LXC image. 

`An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:233
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(453): OC\DB\Connection->connect()`

jandoerntlein, 2025-03-20:
Same issue with LCX image in Proxmox. Cannot activate. 

@neoscaler where can I find the log pointing towards the database issue?

geoW-n, 2025-03-20:
To be more precise:
[Run in Proxmox](https://github.com/nextcloud/nextcloudpi#run-in-proxmox)

- Does not work 1. Using the official release image
- Does work        2. Installer by tteck
Script new location: [NextCloudPi LXC](https://community-scripts.github.io/ProxmoxVE/scripts?id=nextcloudpi)