nextcloud/nextcloudpi #1997: nextcloud check gives error but nextcloud is up&running (but can't be updated)

pkra, 2024-10-25:
I recently noticed that NCP is no longer updating nextcloud itself (I'm still on v28).

The NCP's system info page gives `Nextcloud check: error` however Nextcloud itselfs works just fine - both the web interface and interactions with mobile apps.

While NCP updates just fine, NC updates fail, e.g.,

```bash
$ sudo ncp-update-nc 28.0.11
Running as unit: ncp-update-nc.service
Nextcloud is currently down
Nextcloud update failed (or was installed already).
```

[Since I don't access this installation very often, I don't know exactly how long this has been going on but it may have been since I ran into #1913.]

Any pointers would be greatly appreciated.

<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.55.2
NextcloudPi image    NextCloudPi_03-28-20
OS                   Raspbian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      317G/1.9T
rootfs usage         9.7G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               24min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:139
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1654): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1081): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(261): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(337): OC\DB\Connection->executeQuery()
[... 67 lines cut ...]
```

{"reqId":"SJgGgoZGA21Ii87aZjnH","level":3,"time":"2024-10-25T08:30:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Exception thrown: Doctrine\\DBAL\\Exception","userAgent":"--","version":"28.0.5.1","exception":{"Exception":"Doctrine\\DBAL\\Exception","Message":"Failed to connect to the database: An exception occurred in the driver: could not find driver","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1654,"function":"connect","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1081,"function":"getWrappedConnection","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/Connection.php","line":261,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":337,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":377,"function":"executeQuery","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":280,"function":"execute","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":421,"function":"execute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":187,"function":"loadConfigValues","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":377,"function":"getApps","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/App/AppManager.php","line":128,"function":"getValues","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/App/AppManager.php","line":149,"function":"getInstalledAppsValues","class":"OC\\App\\AppManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":231,"function":"getInstalledApps","class":"OC\\App\\AppManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":90,"function":"getEnabledApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":700,"function":"runInitialRegistration","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":1200,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/DB/Connection.php","Line":139,"CustomMessage":"Exception thrown: Doctrine\\DBAL\\Exception"}}
{"reqId":"TI6aAZZxM6ZI8XUFeiqg","level":3,"time":"2024-10-25T08:35:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Exception thrown: Doctrine\\DBAL\\Exception","userAgent":"--","version":"28.0.5.1","exception":{"Exception":"Doctrine\\DBAL\\Exception","Message":"Failed to connect to the database: An exception occurred in the driver: could not find driver","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1654,"function":"connect","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1081,"function":"getWrappedConnection","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/Connection.php","line":261,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":337,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":377,"function":"executeQuery","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":280,"function":"execute","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":421,"function":"execute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":187,"function":"loadConfigValues","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":377,"function":"getApps","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/App/AppManager.php","line":128,"function":"getValues","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/App/AppManager.php","line":149,"function":"getInstalledAppsValues","class":"OC\\App\\AppManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":231,"function":"getInstalledApps","class":"OC\\App\\AppManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":90,"function":"getEnabledApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":700,"function":"runInitialRegistration","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":1200,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/DB/Connection.php","Line":139,"CustomMessage":"Exception thrown: Doctrine\\DBAL\\Exception"}}
{"reqId":"bPNiQYxmswFdZ1r2pS5S","level":3,"time":"2024-10-25T08:40:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Exception thrown: Doctrine\\DBAL\\Exception","userAgent":"--","version":"28.0.5.1","exception":{"Exception":"Doctrine\\DBAL\\Exception","Mes [... cut ...] nextcloud/lib/base.php","line":700,"function":"runInitialRegistration","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":1200,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/DB/Connection.php","Line":139,"CustomMessage":"Exception thrown: Doctrine\\DBAL\\Exception"}}

```
</details>

theCalcaholic, 2024-11-06:
Your database is down (from the logs it actually looks like a normal shutdown). Can you try running `sudo systemctl start mariadb`?

FadeFx, 2024-11-07:
I doubt mariadb is down, as Nextcloud is working.
Can you try to set an empty mariadb root password?

https://www.tecmint.com/change-mysql-mariadb-root-password/?origin=serp_auto

Atention, don't keep the empty password, make backups of all changed files.

You would need to change the database password in /var/www/nextcloud/config/config.php 
and (probably not, as I think ncp is not able to read it currently) in /root/.my.cnf

After that I had been able to update nextcloud.

Remember to set your previous password again and restore the backuped files.

pkra, 2024-11-07:
@FadeFx thanks for the suggestion.

@theCalcaholic would you suggest the same?

theCalcaholic, 2024-11-07:
Oh right, I might need to fix the report script (it probably uses the wrong source for the db logs).

I'm wondering: Did you manually install a PHP version?

Can you check if php<your-php-version>-mysql is installed? (By default <your-php-version> would be 8.3)

pkra, 2024-11-07:
> I'm wondering: Did you manually install a PHP version?

Hm. I think I tried once (but failed) a few years ago (before debian included php 8). I'm not sure if I also did it while dealing with #1913. 

> Can you check if php-mysql is installed? (By default would be 8.3)

Indeed, it's not. Just to double check: I should install php-mysql, correct?

theCalcaholic, 2024-11-07:
You likely need php8.3-mysql, not php-mysql - unless you tried to manually upgrade php which would have broken NextcloudPi

pkra, 2024-11-07:
> You likely need php8.3-mysql, not php-mysql - unless you tried to manually upgrade php which would have broken NextcloudPi

I installed php8.3-mysql but ncp-update-nc is still failing the same as before. 

I didn't do anything manually after successfully trying the advice of #1913. I'm guessing that does not count as "manual upgrade". I might just have to start fresh on that particular instance.

Do you think it's worth trying the suggestion from https://github.com/nextcloud/nextcloudpi/issues/1997#issuecomment-2461740038 ? 

In any case, thank you again for your help and this wonderful project, @theCalcaholic.

theCalcaholic, 2024-11-07:
@pkra I'm afraid that won't help (and is a security risk).

The error from the Nextcloud Logs clearly indicates a missing mysql driver for PHP (which is installed with the php*-mysql package).

You could try a reboot and then double check what versions of php you have installed on your systemb(try `sudo dpkg -l | grep php`)

pkra, 2024-11-12:
@theCalcaholic thanks and sorry for the slow response.

I tried a reboot (several times at various points).

Here's the result of the grep

<details><summary> output from  <code>sudo dpkg -l | grep php </code>

</summary>

```
rc  libapache2-mod-php7.4               1:7.4.33-10+0~20240422.92+debian12~1.gbp6bc7ef armhf        server-side, HTML-embedded scripting language (Apache 2 module)
rc  libapache2-mod-php8.0               8.0.1-1                                        armhf        server-side, HTML-embedded scripting language (Apache 2 module)
ii  php-common                          2:95+0~20240927.54+debian12~1.gbpe0084c        all          Common files for PHP packages
rc  php-igbinary                        3.2.15-1+0~20231202.49+debian12~1.gbpe9ffe8    armhf        igbinary PHP serializer
ii  php-imagick                         3.7.0-5+0~20240704.44+debian12~1.gbp3f7f5c     armhf        Provides a wrapper to the ImageMagick library
rc  php-redis                           6.0.2-1+0~20231128.56+debian12~1.gbp9f3071     armhf        PHP extension for interfacing with Redis
rc  php7.3-common                       7.3.31-1~deb10u1                               armhf        documentation, examples and common module for PHP
rc  php7.3-readline                     7.3.31-1~deb10u1                               armhf        readline module for PHP
ii  php7.4                              1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac all          server-side, HTML-embedded scripting language (metapackage)
ii  php7.4-bcmath                       1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        Bcmath module for PHP
ii  php7.4-bz2                          1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        bzip2 module for PHP
ii  php7.4-cli                          1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        command-line interpreter for the PHP scripting language
ii  php7.4-common                       1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        documentation, examples and common module for PHP
ii  php7.4-curl                         1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        CURL module for PHP
ii  php7.4-fpm                          1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        server-side, HTML-embedded scripting language (FPM-CGI binary)
ii  php7.4-gd                           1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        GD module for PHP
ii  php7.4-gmp                          1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        GMP module for PHP
ii  php7.4-igbinary                     3.2.16-1+0~20240830.50+debian12~1.gbp150a52    armhf        igbinary PHP serializer
ii  php7.4-intl                         1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        Internationalisation module for PHP
ii  php7.4-json                         1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        JSON module for PHP
ii  php7.4-ldap                         1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        LDAP module for PHP
ii  php7.4-mbstring                     1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        MBSTRING module for PHP
ii  php7.4-mysql                        1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        MySQL module for PHP
ii  php7.4-opcache                      1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        Zend OpCache module for PHP
ii  php7.4-readline                     1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        readline module for PHP
ii  php7.4-redis                        6.0.2-1+0~20231128.56+debian12~1.gbp9f3071     armhf        PHP extension for interfacing with Redis
ii  php7.4-xml                          1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        DOM, SimpleXML, XML, and XSL module for PHP
ii  php7.4-zip                          1:7.4.33-15+0~20240927.95+debian12~1.gbp1b3cac armhf        Zip module for PHP
rc  php8.0-bcmath                       8.0.1-1                                        armhf        Bcmath module for PHP
rc  php8.0-bz2                          8.0.1-1                                        armhf        bzip2 module for PHP
rc  php8.0-cli                          8.0.1-1                                        armhf        command-line interpreter for the PHP scripting language
rc  php8.0-common                       8.0.1-1                                        armhf        documentation, examples and common module for PHP
rc  php8.0-curl                         8.0.1-1                                        armhf        CURL module for PHP
rc  php8.0-fpm                          8.0.1-1                                        armhf        server-side, HTML-embedded scripting language (FPM-CGI binary)
rc  php8.0-gd                           8.0.1-1                                        armhf        GD module for PHP
rc  php8.0-gmp                          8.0.1-1                                        armhf        GMP module for PHP
rc  php8.0-imap                         8.0.1-1                                        armhf        IMAP module for PHP
rc  php8.0-intl                         8.0.1-1                                        armhf        Internationalisation module for PHP
rc  php8.0-ldap                         8.0.1-1                                        armhf        LDAP module for PHP
[... cut ...]
ii  php8.3-phpdbg                       8.3.13-1+0~20241104.46+debian12~1.gbpaaf4dc    armhf        server-side, HTML-embedded scripting language (PHPDBG binary)
ii  php8.3-readline                     8.3.13-1+0~20241104.46+debian12~1.gbpaaf4dc    armhf        readline module for PHP
rc  php8.3-redis                        6.0.2-1+0~20231128.56+debian12~1.gbp9f3071     armhf        PHP extension for interfacing with Redis
```
</details>

It seems to me there are too many versions around and I should probably clean up?

theCalcaholic, 2024-11-12:
Maybe try the following commands first (and then reboot):

```bash
sudo update-alternatives --set php "$(which php8.3)"
sudo a2dismod php7.3
sudo a2dismod php7.4
sudo a2dismod php8.0
sudo a2dismod php8.1
sudo a2dismod php8.2
sudo a2enmod php8.3
sudo service apache2 restart
```

If that works, you can afterwards remove all unneeded php versions:

```bash
sudo apt remove php7.* php8.{0..2}
```

theCalcaholic, 2024-11-12:
Maybe try the following commands first (and then reboot):

```bash
sudo update-alternatives --set php "$(which php8.3)"
sudo a2dismod php7.4
sudo a2dismod php8.1
sudo a2enmod php8.3
sudo service apache2 restart
```

If that works, you can afterwards remove all unneeded php versions:

```bash
sudo apt remove php7.4* php8.1*
```

pkra, 2024-11-14:
Thanks for taking the time, @theCalcaholic. I didn't think I did any manual installations but perhaps I misunderstand what counts as "manual". 

Anyway, I think I've finally broken this system completely. Time to see if there's a viable backup around.

Thank you again for your patient help and this wonderful project.

closed 2024-11-14 by pkra

theCalcaholic, 2024-11-14:
@pkra With manual, I mean "not using the NCP interface". NCP is designed to work filly automated if you don't customize it using the terminal. If you do, you are at risk of breaking things (it's not wrong to take that risk, but it should be a conscious decision).

pkra, 2024-11-14:
@theCalcaholic thanks for clarifying. [As I wrote, I had to follow the instructions in #1913 (which would be "manual" by that definition), so I wasn't sure if that counted. Thanks again.]