nextcloud/nextcloudpi #1998: Trying to access array offset on false

jlfrucot, 2024-10-30:
Many same errors (about 1000 a day)  since upgrade from 29.0.8 to 30.0.1
```

[PHP] Erreur: Trying to access array offset on false at /var/www/nextcloud/apps/settings/lib/SetupChecks/CheckServerResponseTrait.php#97
	GET /settings/ajax/checksetup
	de 192.168.1.254 par frucot à 30 oct. 2024, 16:36:42
```

**System information**

```

NextcloudPi version  v1.55.2
NextcloudPi image    NextCloudPi_11-27-20
OS                   Debian GNU/Linux 11. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/ExternalDrive/ncdata/data/
data in SD           no
data filesystem      btrfs
data disk usage      278G/932G
rootfs usage         9,8G/235G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    30.0.1.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "1": "192.168.1.100",
            "20": "lalbenque.duckdns.org",
            "21": "lalbenque-maison.freeboxos.fr",
            "11": "2a01:e0a:c32:5340::5cd1:e42a",
            "14": "nextcloudpi",
            "3": "nextcloudpi"
        },
        "default_phone_region": "FR",
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "30.0.1.2",
        "overwrite.cli.url": "https:\/\/nextcloudpi\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
[... 111 lines cut ...]
```

{"reqId":"ZyJSK7mOLa5HNZfkjXB37gAAAAY","level":3,"time":"2024-10-30T15:35:08+00:00","remoteAddr":"192.168.1.254","user":"frucot","app":"core","method":"GET","url":"/apps/polls/poll/2/options?time=1730302507431","message":"More than 30 characters for an output column name are not allowed on Oracle.","userAgent":"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:131.0) Gecko/20100101 Firefox/131.0","version":"30.0.1.2","exception":{"Exception":"Doctrine\\DBAL\\Query\\QueryException","Message":"More than 30 characters for an output column name are not allowed on Oracle.","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":284,"function":"prepareForExecute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/public/AppFramework/Db/QBMapper.php","line":253,"function":"executeQuery","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/public/AppFramework/Db/QBMapper.php","line":359,"function":"findOneQuery","class":"OCP\\AppFramework\\Db\\QBMapper","type":"->"},{"file":"/var/www/nextcloud/apps/polls/lib/Db/PollMapper.php","line":65,"function":"findEntity","class":"OCP\\AppFramework\\Db\\QBMapper","type":"->"},{"file":"/var/www/nextcloud/apps/polls/lib/Service/OptionService.php","line":61,"function":"find","class":"OCA\\Polls\\Db\\PollMapper","type":"->"},{"file":"/var/www/nextcloud/apps/polls/lib/Controller/OptionController.php","line":37,"function":"list","class":"OCA\\Polls\\Service\\OptionService","type":"->"},{"file":"/var/www/nextcloud/apps/polls/lib/Controller/BaseController.php","line":39,"function":"OCA\\Polls\\Controller\\{closure}","class":"OCA\\Polls\\Controller\\OptionController","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/apps/polls/lib/Controller/OptionController.php","line":36,"function":"response","class":"OCA\\Polls\\Controller\\BaseController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":208,"function":"list","class":"OCA\\Polls\\Controller\\OptionController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":114,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":161,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":302,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1001,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":24,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","Line":217,"message":"More than 30 characters for an output column name are not allowed on Oracle.","query":"SELECT `polls_polls`.*, (SELECT COUNT(`user_vote_sub`.`vote_answer`) FROM `*PREFIX*polls_votes` `user_vote_sub` WHERE (`user_vote_sub`.`poll_id` = `polls_polls`.`id`) AND (`user_vote_sub`.`user_id` = :dcValue1)) AS `current_user_count_votes`, (SELECT COUNT(`user_vote_sub`.`vote_answer`) FROM `*PREFIX*polls_votes` `user_vote_sub` WHERE (`user_vote_sub`.`poll_id` = `polls_polls`.`id`) AND (`user_vote_sub`.`user_id` = :dcValue1) AND (`user_vote_sub`.`vote_answer` = :dcValue2)) AS `current_user_count_votes_yes`, (SELECT COUNT(`user_vote_sub`.`vote_answer`) FROM `*PREFIX*polls_votes` `user_vote_sub` LEFT JOIN `*PREFIX*polls_options` `vote_options_join` ON (`vote_options_join`.`poll_id` = `user_vote_sub`.`poll_id`) AND (`vote_options_join`.`poll_option_text` = `user_vote_sub`.`vote_option_text`) AND (`vote_options_join`.`deleted` = '0') WHERE (`user_vote_sub`.`poll_id` = `polls_polls`.`id`) AND (`user_vote_sub`.`user_id` = :dcValue1) AND (`vote_options_join`.`id` IS NULL)) AS `current_user_count_orphaned_votes`, coalesce(MAX(options.timestamp), :dcValue3) AS max_date, coalesce(MIN(options.timestamp), :dcValue4) AS min_date, COUNT(`options`.`id`) AS `count_options`, coalesce(user_shares.type, '') AS user_role, `user_shares`.`locked` AS `is_current_user_locked`, coalesce(user_shares.token, '') AS share_token, group_concat(distinct group_shares.user_id SEPARATOR ',') AS group_shares FROM `*PREFIX*polls_polls` `polls_polls` LEFT JOIN `*PREFIX*polls_options` `options` ON (`polls_polls`.`id` = `options`.`poll_id`) AND (`options`.`deleted` = '0') LEFT JOIN `*PREFIX*polls_share` `user_shares` ON (`polls_polls`.`id` = `user_shares`.`poll_id`) AND (`user_shares`.`user_id` = :dcValue5) AND (`user_shares`.`deleted` = '0') LEFT JOIN `*PREFIX*polls_share` `group_shares` ON (`polls_polls`.`id` = `group_shares`.`poll_id`) AND (`group_shares`.`type` = 'group') AND (`group_shares`.`deleted` = '0') WHERE `polls_polls`.`id` = :dcValue6 GROUP BY `polls_polls`.`id`, `user_shares`.`type`, `user_shares`.`locked`, `user_shares`.`token`","columns":["current_user_count_orphaned_votes"],"exception":{},"CustomMessage":"More than 30 characters for an output column name are not allowed on Oracle."}}
{"reqId":"ZyJSK63yy09evmqy0X7I2AAAAAM","level":3,"time":"2024-10-30T15:35:08+00:00","remoteAddr":"192.168.1.254","user":"frucot","app":"core","method":"GET","url":"/apps/polls/poll/2/subscription","message":"More than 30 characters for an output column name are not allowed on Oracle.","userAgent":"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:131.0) Gecko/20100101 Firefox/131.0","version":"30.0.1.2","exception":{"Exception":"Doctrine\\DBAL\\Query\\QueryException","Message":"More than 30 characters for an output column name are not allowed on Oracle.","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":284,"function":"prepareForExecute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/public/AppFramework/Db/QBMapper.php","line":253,"function":"executeQuery","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/public/AppFramework/Db/QBMapper.php","line":359,"function":"findOneQuery","class":"OCP\\AppFramework\\Db\\QBMapper","type":"->"},{"file":"/var/www/nextcloud/apps/polls/lib/Db/PollMapper.php","line":65,"function":"findEntity","class":"OC [... cut ...] ","data":{"app":"PHP"}}
{"reqId":"ZyJSZSDusm-IwQ6OcPvtOwAAAAU","level":3,"time":"2024-10-30T15:36:42+00:00","remoteAddr":"192.168.1.254","user":"frucot","app":"PHP","method":"GET","url":"/settings/ajax/checksetup","message":"Trying to access array offset on false at /var/www/nextcloud/apps/settings/lib/SetupChecks/CheckServerResponseTrait.php#97","userAgent":"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:131.0) Gecko/20100101 Firefox/131.0","version":"30.0.1.2","data":{"app":"PHP"}}

```
</details>

theCalcaholic, 2024-11-06:
That looks like an upstream (Nextcloud server) issue. Please check if it is fixed after the next nextcloud update.

You can either run `ncp-update devel` and then `ncp-update-nc latest` or wait until it's officially released

citizenserious, 2024-11-19:
It seems to be related to https://github.com/nextcloud/server/issues/48820
The issue for me is that I can not remove the ipv6 address, ncp adds one on every restart.
Any ideas how to stop this behavior?

joshtrichards, 2024-11-19:
```
        "trusted_domains": {
            "1": "192.168.1.100",
            "20": "lalbenque.duckdns.org",
            "21": "lalbenque-maison.freeboxos.fr",
            "11": "2a01:e0a:c32:5340::5cd1:e42a",
            "14": "nextcloudpi",
            "3": "nextcloudpi"
```

If ncp is setting the addresses above, if any are IPv6 (such as at position `11`) they need to be enclosed in brackets: https://docs.nextcloud.com/server/latest/admin_manual/configuration_server/config_sample_php_parameters.html#trusted-domains

Looks like it is coming from here:

https://github.com/nextcloud/nextcloudpi/blob/77763af71d57540bdc7edb3a67a7ba5634438f98/bin/nextcloud-domain.sh#L14-L16

Maybe something like:

```
  pub_ipv4="$(curl -4 -m4 icanhazip.com 2>/dev/null)"
  pub_ipv6="$(curl -6 -m4 icanhazip.com 2>/dev/null)"
  [[ "$pub_ipv4"   != "" ]] && ncc config:system:set trusted_domains 11 --value="$pub_ipv4"
  [[ "$pub_ipv6"   != "" ]] && ncc config:system:set trusted_domains 12 --value="[$pub_ipv6]"
```

Caveat: I did not audit all the position assumptions in NCP. 12 may not be a good candidate.

citizenserious, 2024-11-21:
https://github.com/nextcloud/server/issues/49370#issuecomment-2490967063

CarlosCastaneda, 2024-12-28:
After reboot had the error again. Index 11 has been updated with current ipv6 address without brackets (nextcloud_domain.sh).

CarlosCastaneda, 2025-04-03:
One more note: since the prefix of the ipv6 address changes frequently, the current address in config.php entry 11 should take this into account and also be changed, which is not the case. I have found ipv6 addresses on 2 raspberry pi after a restart that no longer exist and consequently lead to the relevant error message in the administration overview.

Codixer, 2025-06-01:
This seems to still be an issue I am facing:

[image]

[image]

[image]