nextcloud/nextcloudpi #2005: Feature request: Support not having root access to Nextcloud data folder

MB-Finski, 2024-11-09:
**Description**  

In an unprivileged Proxmox lxc install if the data folder is an NFS mounted on the proxmox host and passe to the lxc container as a mount bind the root of the lxc won't have root privileges over the files within the mount. This causes the update script to fail even if www-data is owner of the data folder.

It should be fairly straightforward to modify the update script to detect not having root privileges on the mount on the fly and fall back to doing any file manipulations using www-data as the user.

I may look into this if I have time but no promises.

MB-Finski, 2024-11-09:
Hmm.. Seems like the update actually completes successfully at first but the only offending part in the update script is moving the backup file to the nextcloud data directory. I guess we could just run these command with 'sudo -u www-data' in any case since the data dir should be owned by www-data?

korhadris, 2025-05-30:
> Hmm.. Seems like the update actually completes successfully at first but the only offending part in the update script is moving the backup file to the nextcloud data directory. I guess we could just run these commands with 'sudo -u www-data' in any case since the data dir should be owned by www-data?

I just hit the same issue, and came to the same end idea. When the script moves the data, it should perform the actions as the www-data user.