nextcloud/nextcloudpi #2007: error backup

Jeanoo, 2024-11-09:
Hello, 
I am using the latest version of CNP and when I start a backup with nc-backup, I get the following error.
Thank you in advance for your help 

```
Installing template 'ncp-metrics.cfg.sh'...
check free space...
An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:146
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1654): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1081): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(266): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(337): OC\DB\Connection->executeQuery()
#4 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(377): Doctrine\DBAL\Query\QueryBuilder->executeQuery()
#5 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(280): Doctrine\DBAL\Query\QueryBuilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(421): OC\DB\QueryBuilder\QueryBuilder->execute()
#7 /var/www/nextcloud/lib/private/AppConfig.php(187): OC\AppConfig->loadConfigValues()
#8 /var/www/nextcloud/lib/private/AppConfig.php(377): OC\AppConfig->getApps()
#9 /var/www/nextcloud/lib/private/App/AppManager.php(128): OC\AppConfig->getValues()
#10 /var/www/nextcloud/lib/private/App/AppManager.php(149): OC\App\AppManager->getInstalledAppsValues()
#11 /var/www/nextcloud/lib/private/legacy/OC_App.php(231): OC\App\AppManager->getInstalledApps()
#12 /var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php(90): OC_App::getEnabledApps()
#13 /var/www/nextcloud/lib/base.php(700): OC\AppFramework\Bootstrap\Coordinator->runInitialRegistration()
#14 /var/www/nextcloud/lib/base.php(1200): OC::init()
#15 /var/www/nextcloud/console.php(48): require_once('...')
#16 /var/www/nextcloud/occ(11): require_once('...')
#17 {main}Abort...
An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:146
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1654): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1081): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(266): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(337): OC\DB\Connection->executeQuery()
#4 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(377): Doctrine\DBAL\Query\QueryBuilder->executeQuery()
#5 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(280): Doctrine\DBAL\Query\QueryBuilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(421): OC\DB\QueryBuilder\QueryBuilder->execute()
#7 /var/www/nextcloud/lib/private/AppConfig.php(187): OC\AppConfig->loadConfigValues()
#8 /var/www/nextcloud/lib/private/AppConfig.php(377): OC\AppConfig->getApps()
#9 /var/www/nextcloud/lib/private/App/AppManager.php(128): OC\AppConfig->getValues()
#10 /var/www/nextcloud/lib/private/App/AppManager.php(149): OC\App\AppManager->getInstalledAppsValues()
#11 /var/www/nextcloud/lib/private/legacy/OC_App.php(231): OC\App\AppManager->getInstalledApps()
#12 /var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php(90): OC_App::getEnabledApps()
#13 /var/www/nextcloud/lib/base.php(700): OC\AppFramework\Bootstrap\Coordinator->runInitialRegistration()
#14 /var/www/nextcloud/lib/base.php(1200): OC::init()
#15 /var/www/nextcloud/console.php(48): require_once('...')
#16 /var/www/nextcloud/occ(11): require_once('...')
#17 {main}
An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:146
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1654): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1081): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(266): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(337): OC\DB\Connection->executeQuery()
#4 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(377): Doctrine\DBAL\Query\QueryBuilder->executeQuery()
#5 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(280): Doctrine\DBAL\Query\QueryBuilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(421): OC\DB\QueryBuilder\QueryBuilder->execute()
#7 /var/www/nextcloud/lib/private/AppConfig.php(187): OC\AppConfig->loadConfigValues()
#8 /var/www/nextcloud/lib/private/AppConfig.php(377): OC\AppConfig->getApps()
#9 /var/www/nextcloud/lib/private/App/AppManager.php(128): OC\AppConfig->getValues()
#10 /var/www/nextcloud/lib/private/App/AppManager.php(149): OC\App\AppManager->getInstalledAppsValues()
#11 /var/www/nextcloud/lib/private/legacy/OC_App.php(231): OC\App\AppManager->getInstalledApps()
#12 /var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php(90): OC_App::getEnabledApps()
#13 /var/www/nextcloud/lib/base.php(700): OC\AppFramework\Bootstrap\Coordinator->runInitialRegistration()
#14 /var/www/nextcloud/lib/base.php(1200): OC::init()
#15 /var/www/nextcloud/console.php(48): require_once('...')
#16 /var/www/nextcloud/occ(11): require_once('...')
#17 {main}
```

theCalcaholic, 2024-11-14:
This sounds like it could be a duplicate of #1997

Did you at any point install a different PHP version than was provided by the system?

Please share the output of 

```sh
dpkg -l | grep php | grep '^ii'
```

Jeanoo, 2024-11-14:
Thank you for taking charge of my problem. 
No! I haven't installed anything. This is a version installed in August 2024. 

`pi@nextcloudpi:~$ dpkg -l | grep php | grep '^ii'
ii  php-common                     2:94+0~20240205.51+debian12~1.gbp6faa2e      all          Common files for PHP packages
ii  php-imagick                    3.7.0-5+0~20240704.44+debian12~1.gbp3f7f5c   arm64        Provides a wrapper to the ImageMagick library
ii  php8.1                         8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  all          server-side, HTML-embedded scripting language (metapackage)
ii  php8.1-bcmath                  8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        Bcmath module for PHP
ii  php8.1-bz2                     8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        bzip2 module for PHP
ii  php8.1-cli                     8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        command-line interpreter for the PHP scripting language
ii  php8.1-common                  8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        documentation, examples and common module for PHP
ii  php8.1-curl                    8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        CURL module for PHP
ii  php8.1-fpm                     8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        server-side, HTML-embedded scripting language (FPM-CGI binary)
ii  php8.1-gd                      8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        GD module for PHP
ii  php8.1-gmp                     8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        GMP module for PHP
ii  php8.1-igbinary                3.2.15-1+0~20231202.49+debian12~1.gbpe9ffe8  arm64        igbinary PHP serializer
ii  php8.1-intl                    8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        Internationalisation module for PHP
ii  php8.1-ldap                    8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        LDAP module for PHP
ii  php8.1-mbstring                8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        MBSTRING module for PHP
ii  php8.1-mysql                   8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        MySQL module for PHP
ii  php8.1-opcache                 8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        Zend OpCache module for PHP
ii  php8.1-readline                8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        readline module for PHP
ii  php8.1-redis                   6.0.2-1+0~20231128.56+debian12~1.gbp9f3071   arm64        PHP extension for interfacing with Redis
ii  php8.1-xml                     8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        DOM, SimpleXML, XML, and XSL module for PHP
ii  php8.1-zip                     8.1.29-1+0~20240606.65+debian12~1.gbp2ddf4e  arm64        Zip module for PHP
ii  php8.3-cli                     8.3.10-1+0~20240802.37+debian12~1.gbp06babf  arm64        command-line interpreter for the PHP scripting language
ii  php8.3-common                  8.3.10-1+0~20240802.37+debian12~1.gbp06babf  arm64        documentation, examples and common module for PHP
ii  php8.3-imagick                 3.7.0-5+0~20240704.44+debian12~1.gbp3f7f5c   arm64        Provides a wrapper to the ImageMagick library
ii  php8.3-opcache                 8.3.10-1+0~20240802.37+debian12~1.gbp06babf  arm64        Zend OpCache module for PHP
ii  php8.3-phpdbg                  8.3.10-1+0~20240802.37+debian12~1.gbp06babf  arm64        server-side, HTML-embedded scripting language (PHPDBG binary)
ii  php8.3-readline                8.3.10-1+0~20240802.37+debian12~1.gbp06babf  arm64        readline module for PHP
`

**dpkg -l**
`
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                           Version                                      Architecture Description
+++-==============================-============================================-============-==================================================================>
ii  adduser                        3.134                                        all          add and remove users and groups
ii  alsa-utils                     1.2.8-1                                      arm64        Utilities for configuring and using ALSA
ii  apache2                        2.4.62-1~deb12u2                             arm64        Apache HTTP Server
ii  apache2-bin                    2.4.62-1~deb12u2                             arm64        Apache HTTP Server (modules and other binary files)
ii  apache2-data                   2.4.62-1~deb12u2                             all          Apache HTTP Server (common files)
ii  apache2-utils                  2.4.62-1~deb12u2                             arm64        Apache HTTP Server (utility programs for web servers)
ii  apt                            2.6.1                                        arm64        commandline package manager
ii  apt-utils                      2.6.1                                        arm64        package management related utility programs
ii  armbian-bsp-cli-rpi4b-current  24.8.1                                       arm64        Armbian CLI BSP for board 'rpi4b' branch 'current'
ii  avahi-daemon                   0.8-10                                       arm64        Avahi mDNS/DNS-SD daemon
[... cut ...]
 ctrl-DELETE .... ESC-DELETE .... ESC-X ... Delete word under cursor.
 ctrl-U ......... ESC (MS-DOS only) ....... Delete entire line.
 UpArrow ........................ ESC-k ... Retrieve previous command line.
 DownArrow ...................... ESC-j ... Retrieve next command line.
 TAB ...................................... Complete filename & cycle.
 SHIFT-TAB ...................... ESC-TAB   Complete filename & reverse cycle.
 ctrl-L ................................... Complete filename, list all.
`

theCalcaholic, 2024-11-15:
What's the output of `cat /usr/local/etc/ncp.cf` and `php --version`?

You have mixed PHP versions installed - I suspect that you will need to remove php8.3 (but let's wait for the result of the above commands before jumping to any conclusions)

Jeanoo, 2024-11-15:
Thank you again for your availability 

the file ncp.cf  is empty

`php --version
PHP 8.3.10 (cli) (built: Aug  2 2024 16:00:00) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.3.10, Copyright (c) Zend Technologies
    with Zend OPcache v8.3.10, Copyright (c), by Zend Technologies`

Jeanoo, 2024-11-18:
i've just noticed that the version of nextcloudpi 1.55.0 supports php 8.3. should i install nextcloud 29 to solve the problem?
[v1.55.0](https://github.com/nextcloud/nextcloudpi/tree/v1.55.0) (2024-09-07) Support for PHP 8.3 and Nextcloud 29

Jeanoo, 2025-04-13:
an AI gives me this solution to solve my backup problem, can you please give me your opinion if the solution given is viable? 

Here’s the translated response in English:

---

### **Backup Error with `nc-backup` on NextcloudPi**  
The backup error with `nc-backup` on NextcloudPi indicates a database connection issue, specifically related to the missing PHP driver for MySQL/MariaDB (`could not find driver`)[Query]. Below are the steps to resolve this issue:

**Identified Issue**  
The error `Doctrine\DBAL\Exception: Failed to connect to the database: could not find driver` suggests that the required PHP module for database interaction (e.g., `pdo_mysql`) is either not installed or not enabled[Query].

**Possible Solutions**  

1. **Install the PHP Driver for MySQL/MariaDB**  
   - Run the following command via SSH:  
     ```bash
     sudo apt install php-mysql
     ```
   - Restart Apache or PHP-FPM:  
     ```bash
     sudo systemctl restart apache2  # or php-fpm depending on your setup
     ```

2. **Check Loaded PHP Modules**  
   - Use `php -m` to list active modules. Ensure `pdo_mysql` or `mysqli` is present.  
   - If missing, enable it with:  
     ```bash
     sudo phpenmod pdo_mysql
     ```

3. **Verify Database Configuration**  
   - Confirm that the database credentials in Nextcloud’s `config.php` are correct (host, user, password).  
   - Test the connection manually with:  
     ```bash
     mysql -u [user] -p[password] -h [host] [db_name]
     ```

4. **Additional Common Issues**  
   - **Insufficient Disk Space**: Check free space with `df -h` and clean up if needed.  
   - **Database Corruption**: If errors like `Index corrupted` appear, repair the affected table using `mysqlcheck`.  

**After Fixing**  
Rerun `nc-backup` and monitor logs to confirm the operation succeeds. If the issue persists, check detailed logs (`/var/log/ncp.log` or `journalctl -u nextcloud`)[Query].  

---

Let me know if you need further assistance!

---
Réponse de Perplexity: https://www.perplexity.ai/search/erreur-de-sauvegarde-avec-nc-b-2fx8i9StTqe0gB47EIgxnw?utm_source=copy_output