nextcloud/nextcloudpi #2010: Version of mariadb in Nextcloud 30

NickKopernikus, 2024-11-20:
Firstly, I would like to thank you for all your great work on nextcloudpi. Things have been going very well for me lately.

After I successfully updated to the latest version today after a backup, an error about mariadb appears in the overview:

‘MariaDB version ‘10.5.23-MariaDB-0+deb11u1’ detected. For optimal performance, stability and functionality with this version of Nextcloud, MariaDB >= 10.6 and 11.4 <= is recommended.’

Currently everything seems to be running normally. Nevertheless my questions:

- Will this still be updated automatically at some point?
- Does it make sense to do the whole thing manually? (Instructions?)

Thank you very much!

Translated with DeepL.com (free version)

NickKopernikus, 2024-11-20:
This is the output - upgrade canceled:

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libabsl20200923 libaio1 libavif9 libdav1d4 libfmt7 libgav1-0 libpcre2-posix2
  mariadb-server-10.5
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  mariadb-client-10.5 mariadb-client-core-10.5 mariadb-server-core-10.5
The following NEW packages will be installed:
  dconf-cli debsuryorg-archive-keyring libabsl20220623 libavif15 libbluetooth3
  libdaxctl1 libdconf1 libfmt9 libgav1-1 libimagequant0 libmbim-glib4
  libmbim-proxy libmbim-utils libmm-glib0 libndctl6 libndp0 libnm0
  libossp-uuid16 libpcap0.8 libpmem1 libqmi-glib5 libqmi-proxy libqmi-utils
  libqrtr-glib0 libraqm0 librav1e0 libsvtav1enc1 libteamdctl0 liburing2
  libyuv0 mariadb-client mariadb-client-core mariadb-plugin-provider-bzip2
  mariadb-plugin-provider-lz4 mariadb-plugin-provider-lzma
  mariadb-plugin-provider-lzo mariadb-plugin-provider-snappy
  mariadb-server-core modemmanager network-manager ppp python3-pycryptodome
  uuid
The following packages have been kept back:
  exfat-fuse ghostscript libgs9-common libraspberrypi-bin libraspberrypi-dev
  libraspberrypi-doc libraspberrypi0 libsemanage-common nfs-common
  nfs-kernel-server ntfs-3g passwd raspi-utils
The following packages will be upgraded:
  alsa-utils bluez bluez-firmware dphys-swapfile dpkg dpkg-dev
  firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree
  firmware-realtek initramfs-tools initramfs-tools-core kms++-utils libasound2
  libasound2-data libatopology2 libc-bin libc-dev-bin libc-devtools libc-l10n
  libc6 libc6-dbg libc6-dev libcairo2 libdpkg-perl libdrm-common libdrm2
  libgd3 libkms++0 libpam-chksshpwd libpam-modules libpam-modules-bin libpam0g
  libpcre3 libpixman-1-0 libssl3 linux-libc-dev locales mariadb-server
  mariadb-server-10.5 openssl php-common php8.3 php8.3-bcmath php8.3-bz2
  php8.3-cli php8.3-common php8.3-curl php8.3-fpm php8.3-gd php8.3-gmp
  php8.3-intl php8.3-ldap php8.3-mbstring php8.3-mysql php8.3-opcache
  php8.3-readline php8.3-xml php8.3-zip pi-bluetooth raspberrypi-net-mods
  raspberrypi-sys-mods raspi-config raspi-gpio rpi-eeprom rpi-update udisks2
  userconf-pi zlib1g zlib1g-dev
71 upgraded, 43 newly installed, 3 to remove and 13 not upgraded.
Need to get 182 MB of archives.
After this operation, 86.1 MB of additional disk space will be used.

Is it safe to upgrade this way?

Oberlurch3000, 2025-01-11:
I've got the same issue since updating to NC 30.
I'm currently running NC on debian 11 (bullseye) 64bit raspi 4.
Mariadb 10.5.23 seems to be the most up to date version you can get from the debian 11 repositories.
So I could upgrade to debian 12 with ``ncp-dist-upgrade``. But I don't know how easy this process will be and what other problems this would cause.
The other way is to get for example mariadb 11.4 from an external repository like this: ``https://mariadb.org/download/?t=repo-config&d=Debian+11+%22Bullseye%22&v=11.4&r_m=23Media``
Since this version is not really supported for debian 11 I get the following output when running ``apt-get update``:

Get:1 http://security.debian.org/debian-security bullseye-security InRelease [27.2 kB]
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Get:3 http://security.debian.org/debian-security bullseye-security/main armhf Packages [325 kB]
Hit:4 http://deb.debian.org/debian bullseye InRelease
Get:5 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Hit:6 http://packages.azlux.fr/debian buster InRelease
Hit:7 https://mirror.23m.com/mariadb/repo/11.4/debian bullseye InRelease
Hit:8 https://packages.sury.org/php bullseye InRelease
Fetched 396 kB in 2s (200 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
61 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://mirror.23m.com/mariadb/repo/11.4/debian bullseye InRelease' doesn't support architecture 'armhf'

I would like to give it a try after making a backup of my system if i would know what the note at the end means for me.
Does it just say that this version of mariadb wasn't ment to be installed on this system or will it break it?

My NC data and database are stored on a different device than the system itself. So I hope I could go back to the backed up system if it doesn't work. Or will the update of mariadb to version 11.4 change the database so my backup system can't read it anymore?

NickKopernikus, 2025-01-12:
Well - as I updated my setup to a rpi 5, I made a fresh and clean Installation. I Just backed everything up with nextcloudpi and die a restore. With the usage of the latest Image, there was no issue and everything runs now in latest Version. After nearly 5 yeara 24/7 uptime, i think this effort is ok.

So I am sorry, but I cant help you with the issue but wanted to give my Feedback as I initially started the thread...