nextcloud/nextcloudpi #2013: Error while applying update 1.55.3. Exiting...

eggi36, 2024-11-27:
The first time I got an issue with updating ncp.
Version 1.55.2 is currently installed. Updating to 1.55.3 just gives me an error. Doesn't matter if executed via web interface or cli.

`ncp-update`
`Downloading updates`
`Performing updates`
`Cannot load Zend OPcache - it was already loaded`
`Cannot load Zend OPcache - it was already loaded`
`Cannot load Zend OPcache - it was already loaded`
`Error while applying update 1.55.3. Exiting...`

<details>
<summary>NextcloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
NextcloudPi version  v1.55.2
OS                   Debian GNU/Linux 12. 6.1.0-27-amd64 (x86_64)
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      29G/102G
rootfs usage         29G/102G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    30.0.2.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "***REMOVED SENSITIVE VALUE***",
            "1": "192.168.0.29",
            "14": "nextcloud",
            "3": "nextcloud",
            "5": "***REMOVED SENSITIVE VALUE***",
            "6": "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "30.0.2.2",
        "overwrite.cli.url": "https:\/\/nextcloud\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "loglevel": "2",
        "log_type": "file",
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "tls",
        "mail_smtpauth": 1,
        "mail_smtpport": "587",
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "maintenance": false,
        "theme": "",
        "updater.release.channel": "stable",
        "app_install_overwrite": [
            "nextcloudpi",
            "previewgenerator"
        ],
        "htaccess.RewriteBase": "\/",
        "default_phone_region": "DE",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "enabledPreviewProviders": [
            "OC\\Preview\\PNG",
            "OC\\Preview\\JPEG",
            "OC\\Preview\\GIF",
            "OC\\Preview\\BMP",
            "OC\\Preview\\XBitmap",
            "OC\\Preview\\MP3",
            "OC\\Preview\\TXT",
            "OC\\Preview\\MarkDown",
            "OC\\Preview\\OpenDocument",
            "OC\\Preview\\Krita",
            "OC\\Preview\\HEIC",
            "OC\\Preview\\Movie",
            "OC\\Preview\\MKV",
            "OC\\Preview\\MP4",
            "OC\\Preview\\AVI",
            "OC\\Preview\\PDF"
        ],
        "maintenance_window_start": 100
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Nov 27 00:00:00.554187 2024] [ssl:warn] [pid 1127:tid 1127] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Nov 27 00:00:00.554581 2024] [ssl:error] [pid 1127:tid 1127] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=nextcloud / issuer: CN=nextcloud / serial: 93A9AE9E8684783A / notbefore: Apr 27 12:55:56 2019 GMT / notafter: Apr 24 12:55:56 2029 GMT]
[Wed Nov 27 00:00:00.554590 2024] [ssl:error] [pid 1127:tid 1127] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Wed Nov 27 00:00:00.556876 2024] [http2:warn] [pid 1127:tid 1127] AH10034: The mpm module (prefork.c) is not supported by mod_http2. The mpm determines how things are processed in your server. HTTP/2 has more demands in this regard and the currently selected mpm will just not do. This is an advisory warning. Your server will continue to work, but the HTTP/2 protocol will be inactive.
[Wed Nov 27 00:00:00.557638 2024] [mpm_prefork:notice] [pid 1127:tid 1127] AH00163: Apache/2.4.62 (Debian) OpenSSL/3.0.15 configured -- resuming normal operations
[Wed Nov 27 00:00:00.557645 2024] [core:notice] [pid 1127:tid 1127] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2022-03-06 16:40:53 0 [Note] InnoDB: Waiting for purge to start
2022-03-06 16:40:53 0 [Note] InnoDB: 10.3.31 started; log sequence number 14083864134; transaction id 43459910
2022-03-06 16:40:53 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2022-03-06 16:40:53 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-03-06 16:40:53 0 [Note] Server socket created on IP: '127.0.0.1'.
2022-03-06 16:40:53 0 [Note] Reading of all Master_info entries succeeded
2022-03-06 16:40:53 0 [Note] Added new Master_info '' to hash table
2022-03-06 16:40:53 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.31-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2022-03-06 16:40:53 0 [Note] InnoDB: Buffer pool(s) load completed at 220306 16:40:53
2022-03-06 16:49:48 0 [Note] /usr/sbin/mysqld (initiated by: unknown): Normal shutdown
2022-03-06 16:49:48 0 [Note] Event Scheduler: Purging the queue. 0 events
2022-03-06 16:49:48 0 [Note] InnoDB: FTS optimize thread exiting.
2022-03-06 16:49:48 0 [Note] InnoDB: Starting shutdown...
2022-03-06 16:49:48 0 [Note] InnoDB: Dumping buffer pool(s) to /var/lib/mysql/ib_buffer_pool
2022-03-06 16:49:48 0 [Note] InnoDB: Buffer pool(s) dump completed at 220306 16:49:48
2022-03-06 16:49:49 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2022-03-06 16:49:49 0 [Note] InnoDB: Shutdown completed; log sequence number 14084189614; transaction id 43460534
2022-03-06 16:49:49 0 [Note] /usr/sbin/mysqld: Shutdown complete

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{***REMOVED AS IT COMTAINED SENSITIVE INFORMATION***}

```
</details>

eggi36, 2024-12-12:
resolved. Updated successfully by cloning the repository. Had to comment lines 106-109 in run_update_history.sh to let it update successfully.
Ran ncp-update afterwards and everything seems fine now.

closed 2024-12-12 by eggi36

PiBuilder5, 2024-12-14:
where ist the run_update_history.sh file?

eggi36, 2024-12-15:
its in this repository in the root folder.