nextcloud/nextcloudpi #2015: Endless activation

mak163, 2024-12-01:
I have installed NextCloudPi in the LXC Proxmox container. After installation, when I go to the page https://nextcloudpi.local/activate / the system suggests saving passwords and activating NextCloudPi . When I press "Activate", the message "NextCloudPi not yet initialized, trying again in a few seconds ..." is displayed and the circle turns endlessly.

cblt2l, 2024-12-27:
I ran into the same issue trying to install as an LXC on Proxmox. I set it up with 1 cpu core, 1GiB ram and set it to run as a privileged container.

running `journalctl -ef` in the running container, I could see the following lines repeating endlessly.

```
Dec 27 17:42:45 nextcloud sudo[4164]:     root : PWD=/ ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ config:system:get overwrite.cli.url
Dec 27 17:42:45 nextcloud sudo[4164]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Dec 27 17:42:45 nextcloud sudo[4164]: pam_unix(sudo:session): session closed for user www-data
Dec 27 17:42:48 nextcloud sudo[4168]:     root : PWD=/ ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ config:system:get overwrite.cli.url
Dec 27 17:42:48 nextcloud sudo[4168]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Dec 27 17:42:48 nextcloud sudo[4168]: pam_unix(sudo:session): session closed for user www-data
Dec 27 17:42:49 nextcloud sudo[4172]: www-data : PWD=/var/www/ncp-web ; USER=root ; COMMAND=/home/www/ncp-launcher.sh nc-admin
Dec 27 17:42:49 nextcloud sudo[4172]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Dec 27 17:42:49 nextcloud sudo[4191]:     root : PWD=/var/www/ncp-web ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ status
Dec 27 17:42:49 nextcloud sudo[4191]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Dec 27 17:42:49 nextcloud sudo[4191]: pam_unix(sudo:session): session closed for user www-data
Dec 27 17:42:49 nextcloud sudo[4218]:     root : PWD=/var/www/ncp-web ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ user:resetpassword --password-from-env ncp
Dec 27 17:42:49 nextcloud sudo[4218]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
```

Futty87, 2025-01-16:
Hello!
same problem here, in my case the database is probably not running...

`systemctl status mysql
× mariadb.service - MariaDB 10.11.6 database server
     Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2025-01-16 19:11:57 UTC; 1h 41min ago
       Docs: man:mariadbd(8)
             https://mariadb.com/kb/en/library/systemd/
   Main PID: 337 (code=exited, status=1/FAILURE)
     Status: "MariaDB server is down"
        CPU: 129ms

Jan 16 19:11:57 nextcloud systemd[1]: Starting mariadb.service - MariaDB 10.11.6 database server...
Jan 16 19:11:57 nextcloud mariadbd[337]: 2025-01-16 19:11:57 0 [Warning] Can't create test file '/var/lib/mysql/nextcloud.lower-test' (Errcode: 13 "Permission denied")
Jan 16 19:11:57 nextcloud mariadbd[337]: 2025-01-16 19:11:57 0 [Note] Starting MariaDB 10.11.6-MariaDB-0+deb12u1 source revision  as process 337
Jan 16 19:11:57 nextcloud mariadbd[337]: 2025-01-16 19:11:57 0 [ERROR] mariadbd: Can't create/write to file './ddl_recovery.log' (Errcode: 13 "Permission denied")
Jan 16 19:11:57 nextcloud mariadbd[337]: 2025-01-16 19:11:57 0 [ERROR] DDL_LOG: Failed to create ddl log file: ./ddl_recovery.log
Jan 16 19:11:57 nextcloud mariadbd[337]: 2025-01-16 19:11:57 0 [ERROR] Aborting
Jan 16 19:11:57 nextcloud systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 16 19:11:57 nextcloud systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 16 19:11:57 nextcloud systemd[1]: Failed to start mariadb.service - MariaDB 10.11.6 database server.`

Scud0, 2025-01-27:
today still an issue with latest;  NextcloudPi_LXC_x86_v1.55.3.tar.gz. 

installed and pressing activate results in an error in ncp.log
```
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:233
```

Belaial, 2025-02-03:
Mentioned here also,

https://github.com/nextcloud/nextcloudpi/issues/1994

But I'm experiencing the same thing in Proxmox with LXC and "NextcloudPi_LXC_x86_v1.55.3.tar.gz", have not looked into logs and so on since I just wanted to test this out, will keep an eye on how this might progress.

neoscaler, 2025-03-13:
Can confirm issue is still present in latest NextcloudPi_LXC_x86_v1.55.3.tar.gz