nextcloud/nextcloudpi #2016: No WIFI support for Raspberry Pi 5 in release image 1.55.3

mozbius, 2024-12-02:
### PROBLEMS

Wifi  networking is not working with NextcloudPi 1.55.3 on Raspberry Pi 5 with integrated wifi interface.

Troubleshooting steps that reveal the issue can be found at:

https://help.nextcloud.com/t/how-to-make-wifi-work-with-nextcloudpi-on-raspberry-pi-5/211142/15

**System information**

xyzadmin@nextcloudpi:~$ sudo ncp-report
[sudo] password for ncphradmin:
<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.55.3
NextcloudPi image
OS                   Armbian-unofficial 24.8.2 Bookworm \l . 6.6.45-current-bcm2712 (aarch64)
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.8G/29G
rootfs usage         4.8G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    30.0.1.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            end0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               2:03

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "14": "nextcloudpi",
[... 86 lines cut ...]
```

{"reqId":"NKokxs3cRxiORAjeXMYc","level":2,"time":"2024-12-02T11:54:27+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"issue while running UpdateSingleMetadata","userAgent":"--","version":"30.0.1.2","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":59,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/Memcache/WithLocalCache.php","line":28,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/User/Manager.php","line":131,"function":"get","class":"OC\\Memcache\\WithLocalCache","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/Root.php","line":342,"function":"get","class":"OC\\User\\Manager","type":"->"},{"function":"getUserFolder","class":"OC\\Files\\Node\\Root","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyFolder.php","line":64,"function":"call_user_func_array"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyRoot.php","line":38,"function":"__call","class":"OC\\Files\\Node\\LazyFolder","type":"->"},{"file":"/var/www/nextcloud/lib/private/FilesMetadata/Job/UpdateSingleMetadata.php","line":41,"function":"getUserFolder","class":"OC\\Files\\Node\\LazyRoot","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/Job.php","line":61,"function":"run","class":"OC\\FilesMetadata\\Job\\UpdateSingleMetadata","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/QueuedJob.php","line":43,"function":"start","class":"OCP\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/QueuedJob.php","line":29,"function":"start","class":"OCP\\BackgroundJob\\QueuedJob","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":162,"function":"execute","class":"OCP\\BackgroundJob\\QueuedJob","type":"->"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":59,"message":"issue while running UpdateSingleMetadata","exception":{},"userId":"ncp","fileId":48,"CustomMessage":"issue while running UpdateSingleMetadata"}}
{"reqId":"NKokxs3cRxiORAjeXMYc","level":2,"time":"2024-12-02T11:54:27+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"issue while running UpdateSingleMetadata","userAgent":"--","version":"30.0.1.2","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":59,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/Memcache/WithLocalCache.php","line":28,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/User/Manager.php","line":131,"function":"get","class":"OC\\Memcache\\WithLocalCache","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/Root.php","line":342,"function":"get","class":"OC\\User\\Manager","type":"->"},{"function":"getUserFolder","class":"OC\\Files\\Node\\Root","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyFolder.php","line":64,"function":"call_user_func_array"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyRoot.php","line":38,"function":"__call","class":"OC\\Files\\Node\\LazyFolder","type":"->"},{"file":"/var/www/nextcloud/lib/private/FilesMetadata/Job/UpdateSingleMetadata.php","line":41,"function":"getUserFolder","class":"OC\\Files\\Node\\LazyRoot","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/Job.php","line":61,"function":"run","class":"OC\\FilesMetadata\\Job\\UpdateSingleMetadata","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/QueuedJob.php","line":43,"function":"start","class":"OCP\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/QueuedJob.php","line":29,"function":"start","class":"OCP\\BackgroundJob\\QueuedJob","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":162,"function":"execute","class":"OCP\\BackgroundJob\\QueuedJob","type":"->"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":59,"message":"issue while running UpdateSingleMetadata","exception":{},"userId":"ncp","fileId":49,"CustomMessage":"issue while running UpdateSingleMetadata"}}
{"reqId":"NKokxs3cRxiORAjeXMYc","level":2,"time":"2024-12-02T11:54:27+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"issue while running UpdateSingleMetadata","userAgent":"--","version":"30.0.1.2","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":59,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/Memcache/WithLocalCache.php","line":28,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/User/Manager.php","line":131,"function":"get","class":"OC\\Memcache\\WithLocalCache","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/Root.php","line":342,"function":"get","class":"OC\\User\\Manager","type":"->"},{"function":"getUserFolder","class":"OC\\Files\\Node\\Root","type":"->"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyFolder.php","line":64,"function":"call_user_func_array"},{"file":"/var/www/nextcloud/lib/private/Files/Node/LazyRoot.php","line":38,"function":"__call","class":"OC\\Files\\Node\\LazyFolder","type":"->"},{"file":"/var/www/nextcloud/lib/private/FilesMetadata/Job/UpdateSingleMetadata.php","line":41,"function":"getUserFolder","class":"OC\\Files\\Node\\LazyRoot","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/Job.php","line":61,"function":"run","class":"OC\\FilesMetadata\\Job\\UpdateSingleMetadata","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/QueuedJob.php","line":43,"function":"start","class":"OCP\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/lib/public/BackgroundJob/QueuedJob.php","line":29,"function":"start","class":"OCP\\BackgroundJob\\QueuedJob","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":162,"function":"execute","class":"OCP\\BackgroundJob\\QueuedJob","type":"->"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":59,"message":"issue while running UpdateSingleMetadata","exception":{},"userId":"ncp","fileId":50,"CustomMessage":"iss [... cut ...]
{"reqId":"tv3Ul5JcJuSDnaakZTpT","level":2,"time":"2024-12-02T12:56:03+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /opt/ncdata/data/tmp is not present or writable","userAgent":"--","version":"30.0.1.2","data":[]}
{"reqId":"OIpdPhpOD1xyOCGF3pQa","level":2,"time":"2024-12-02T12:56:03+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /opt/ncdata/data/tmp is not present or writable","userAgent":"--","version":"30.0.1.2","data":[]}

```
</details>

AaronB2019, 2024-12-03:
Yes, I'm having the same issue too, I saw some people having success with installing raspian-config and network-manager but it doesn't work for me.

dwprod38, 2025-06-20:
See my post which solve the same issue with armbian + RPi5 #2055

mozbius, 2025-06-20:
Thank you. I will try that!!