nextcloud/nextcloudpi #2017: Won't boot after update (due to upgraded kernel "6.10.11+bpo-arm64-16k" whereas lib/mobules are still for "6.6.45-current-bcm2712")

fhamborg, 2024-12-04:
Every time the system does an udpate (automatically) or I perform one manually (`apt update && apt upgrade`), it won't be able to boot next time I restart it. Thus far, I always ended up `dd`ing the boot partition from a fresh nextcloudpi image to my USB stick and would boot again. This time I finally was able to get to the direct cause of the system not being able to boot, which is that a new kernel version is installed:

```
6.10.11+bpo-arm64-16k
```

compared to `6.6.45-current-bcm2712` on the fresh nextcloudpi image. While this not bad per se, I noticed that the other files were not updated correctly, e.g., note how `lib/modules` is still on 6.6.45 (whereas the kernel on the old system was already upgraded to 6.10.11):

```
ls /mnt/oldroot/lib/modules
6.6.45-current-bcm2712
```

Now I think there might be two potential issues here: 

1) Is the kernel upgrade even expected or should a "normal" nextcloudpi installation stick with 6.6.45?
2) If it's expected to upgrade, I think the other files should upgrade, too. If not, why would it upgrade to 6.10.11?

FYI, the diff between the old (broken) initrd.img and the fresh one (note that some files related to cryptsetup are added to the old/broken setup (perhaps because I enabled encryption in the old setup)

```
diff -r /tmp/initrd-new /tmp/initrd-old
Only in /tmp/initrd-old/bin: cryptroot-unlock
Only in /tmp/initrd-old/bin: kbd_mode
Only in /tmp/initrd-old/bin: loadkeys
Only in /tmp/initrd-old/bin: setupcon
Binary files /tmp/initrd-new/bin/udevadm and /tmp/initrd-old/bin/udevadm differ
Only in /tmp/initrd-old: cryptroot
Only in /tmp/initrd-old/etc: console-setup
Binary files /tmp/initrd-new/etc/ld.so.cache and /tmp/initrd-old/etc/ld.so.cache differ
Only in /tmp/initrd-old/etc: motd
Binary files /tmp/initrd-new/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1 and /tmp/initrd-old/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1 differ
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libargon2.so.1
Binary files /tmp/initrd-new/lib/aarch64-linux-gnu/libcrypto.so.3 and /tmp/initrd-old/lib/aarch64-linux-gnu/libcrypto.so.3 differ
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libcryptsetup.so.12
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libcryptsetup.so.12.9.0
Binary files /tmp/initrd-new/lib/aarch64-linux-gnu/libc.so.6 and /tmp/initrd-old/lib/aarch64-linux-gnu/libc.so.6 differ
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libgcc_s.so.1
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libjson-c.so.5
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libjson-c.so.5.2.0
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libpopt.so.0
Only in /tmp/initrd-old/lib/aarch64-linux-gnu: libpopt.so.0.0.2
Binary files /tmp/initrd-new/lib/aarch64-linux-gnu/libresolv.so.2 and /tmp/initrd-old/lib/aarch64-linux-gnu/libresolv.so.2 differ
Binary files /tmp/initrd-new/lib/aarch64-linux-gnu/libudev.so.1 and /tmp/initrd-old/lib/aarch64-linux-gnu/libudev.so.1 differ
Binary files /tmp/initrd-new/lib/aarch64-linux-gnu/libudev.so.1.7.5 and /tmp/initrd-old/lib/aarch64-linux-gnu/libudev.so.1.7.5 differ
Only in /tmp/initrd-old/lib: cryptsetup
Binary files /tmp/initrd-new/lib/ld-linux-aarch64.so.1 and /tmp/initrd-old/lib/ld-linux-aarch64.so.1 differ
Only in /tmp/initrd-old/lib/modules: 6.10.11+bpo-arm64-16k
Only in /tmp/initrd-new/lib/modules: 6.6.45-current-bcm2712
Binary files /tmp/initrd-new/lib/systemd/systemd-udevd and /tmp/initrd-old/lib/systemd/systemd-udevd differ
Binary files /tmp/initrd-new/lib/udev/ata_id and /tmp/initrd-old/lib/udev/ata_id differ
Binary files /tmp/initrd-new/lib/udev/scsi_id and /tmp/initrd-old/lib/udev/scsi_id differ
Only in /tmp/initrd-old/sbin: cryptsetup
Only in /tmp/initrd-old/scripts: local-block
Only in /tmp/initrd-old/scripts: local-bottom
Only in /tmp/initrd-old/scripts: local-top
Only in /tmp/initrd-old/usr/bin: cryptroot-unlock
Only in /tmp/initrd-old/usr/bin: kbd_mode
Only in /tmp/initrd-old/usr/bin: loadkeys
Only in /tmp/initrd-old/usr/bin: setupcon
Binary files /tmp/initrd-new/usr/bin/udevadm and /tmp/initrd-old/usr/bin/udevadm differ
Binary files /tmp/initrd-new/usr/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1 and /tmp/initrd-old/usr/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1 differ
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libargon2.so.1
Binary files /tmp/initrd-new/usr/lib/aarch64-linux-gnu/libcrypto.so.3 and /tmp/initrd-old/usr/lib/aarch64-linux-gnu/libcrypto.so.3 differ
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libcryptsetup.so.12
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libcryptsetup.so.12.9.0
Binary files /tmp/initrd-new/usr/lib/aarch64-linux-gnu/libc.so.6 and /tmp/initrd-old/usr/lib/aarch64-linux-gnu/libc.so.6 differ
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libgcc_s.so.1
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libjson-c.so.5
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libjson-c.so.5.2.0
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libpopt.so.0
Only in /tmp/initrd-old/usr/lib/aarch64-linux-gnu: libpopt.so.0.0.2
Binary files /tmp/initrd-new/usr/lib/aarch64-linux-gnu/libresolv.so.2 and /tmp/initrd-old/usr/lib/aarch64-linux-gnu/libresolv.so.2 differ
Binary files /tmp/initrd-new/usr/lib/aarch64-linux-gnu/libudev.so.1 and /tmp/initrd-old/usr/lib/aarch64-linux-gnu/libudev.so.1 differ
Binary files /tmp/initrd-new/usr/lib/aarch64-linux-gnu/libudev.so.1.7.5 and /tmp/initrd-old/usr/lib/aarch64-linux-gnu/libudev.so.1.7.5 differ
Only in /tmp/initrd-old/usr/lib: cryptsetup
Binary files /tmp/initrd-new/usr/lib/ld-linux-aarch64.so.1 and /tmp/initrd-old/usr/lib/ld-linux-aarch64.so.1 differ
Only in /tmp/initrd-old/usr/lib/modules: 6.10.11+bpo-arm64-16k
Only in /tmp/initrd-new/usr/lib/modules: 6.6.45-current-bcm2712
Binary files /tmp/initrd-new/usr/lib/systemd/systemd-udevd and /tmp/initrd-old/usr/lib/systemd/systemd-udevd differ
Binary files /tmp/initrd-new/usr/lib/udev/ata_id and /tmp/initrd-old/usr/lib/udev/ata_id differ
Binary files /tmp/initrd-new/usr/lib/udev/scsi_id and /tmp/initrd-old/usr/lib/udev/scsi_id differ
Only in /tmp/initrd-old/usr/sbin: cryptsetup
```

geoW-n, 2024-12-05:
> I perform one manually (`apt update && apt upgrade`), it won't be able to boot next time I restart it.

I assume you are running an armbian based image on a pi, then you should consider uüdating manual by **armbian-upgrade**. It took my test-pi to  v24.11.1 for Raspberry Pi 4 running Armbian Linux 6.6.63-current-bcm2711.

fhamborg, 2024-12-05:
I thought the proposed way to update was ncp-update?

geoW-n, 2024-12-05:
> to update was ncp-update

To update nextcloudpi, yes.

> I perform one manually (`apt update && apt upgrade`)

Therefore my hint to use armbian-upgrade.

fhamborg, 2024-12-05:
Alright, so that's interesting. I started on the freshly recovered system (where I had recovered it by `dd`ing the boot/firmare partition from the latest ncp image to the usb stick - nothing else) an update using `armbian-upgrade` and it seems the kernel is being updated (or rather a new one installed additionally):

```
The following NEW packages will be installed:
  linux-image-6.11.5+bpo-arm64-16k
The following packages will be upgraded:
  containerd.io linux-image-arm64-16k php8.3 php8.3-bcmath php8.3-bz2 php8.3-cli php8.3-common php8.3-curl php8.3-fpm php8.3-gd php8.3-gmp php8.3-igbinary php8.3-intl php8.3-ldap php8.3-mbstring php8.3-mysql
  php8.3-opcache php8.3-readline php8.3-redis php8.3-xml php8.3-zip tzdata
22 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
```

This is exactly the one that is also present when I see that the system doesn't boot. But then the updater doesn't finish because of this:

```
Errors were encountered while processing:
 linux-image-6.11.5+bpo-arm64-16k
 linux-image-arm64-16k
E: Sub-process /usr/bin/dpkg returned an error code (1)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up linux-image-6.11.5+bpo-arm64-16k (6.11.5-1~bpo12+1) ...
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.11.5+bpo-arm64-16k
/etc/kernel/postinst.d/zzz-copy-new-files:
cp: cannot stat '/usr/lib/linux-image-6.11.5+bpo-arm64-16k//overlays': No such file or directory
run-parts: /etc/kernel/postinst.d/zzz-copy-new-files exited with return code 1
dpkg: error processing package linux-image-6.11.5+bpo-arm64-16k (--configure):
 installed linux-image-6.11.5+bpo-arm64-16k package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-image-arm64-16k:
 linux-image-arm64-16k depends on linux-image-6.11.5+bpo-arm64-16k (= 6.11.5-1~bpo12+1); however:
  Package linux-image-6.11.5+bpo-arm64-16k is not configured yet.

dpkg: error processing package linux-image-arm64-16k (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 linux-image-6.11.5+bpo-arm64-16k
 linux-image-arm64-16k
^[[AE: Sub-process /usr/bin/dpkg returned an error code (1)
```

fhamborg, 2024-12-05:
would you be so kind and also post the contents of /boot/firmware ?

geoW-n, 2024-12-05:
> the contents of /boot/firmware

```
pi@ncpi4:~$ ls -la /boot/firmware
insgesamt 60277
drwxr-xr-x 4 root root     4096  1. Jan 1970   .
drwxr-xr-x 4 root root     4096 30. Nov 21:44  ..
-rwxr-xr-x 1 root root    32519 30. Nov 06:04  bcm2710-rpi-2-b.dtb
-rwxr-xr-x 1 root root    34711 30. Nov 06:04  bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root    35346 30. Nov 06:04  bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root    32298 30. Nov 06:04  bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root    33688 30. Nov 06:04  bcm2710-rpi-zero-2.dtb
-rwxr-xr-x 1 root root    33688 30. Nov 06:04  bcm2710-rpi-zero-2-w.dtb
-rwxr-xr-x 1 root root    56160 30. Nov 06:04  bcm2711-rpi-400.dtb
-rwxr-xr-x 1 root root    56156 30. Nov 06:04  bcm2711-rpi-4-b.dtb
-rwxr-xr-x 1 root root    56640 30. Nov 06:04  bcm2711-rpi-cm4.dtb
-rwxr-xr-x 1 root root    38405 30. Nov 06:04  bcm2711-rpi-cm4-io.dtb
-rwxr-xr-x 1 root root    53443 30. Nov 06:04  bcm2711-rpi-cm4s.dtb
-rwxr-xr-x 1 root root    81455 30. Nov 06:04  bcm2712d0-rpi-5-b.dtb
-rwxr-xr-x 1 root root    81407 30. Nov 06:04  bcm2712-rpi-500.dtb
-rwxr-xr-x 1 root root    81451 30. Nov 06:04  bcm2712-rpi-5-b.dtb
-rwxr-xr-x 1 root root    81807 30. Nov 06:04  bcm2712-rpi-cm5-cm4io.dtb
-rwxr-xr-x 1 root root    81869 30. Nov 06:04  bcm2712-rpi-cm5-cm5io.dtb
-rwxr-xr-x 1 root root    81848 30. Nov 06:04  bcm2712-rpi-cm5l-cm4io.dtb
-rwxr-xr-x 1 root root    81910 30. Nov 06:04  bcm2712-rpi-cm5l-cm5io.dtb
-rwxr-xr-x 1 root root    20769 30. Nov 06:04  bcm2837-rpi-3-a-plus.dtb
-rwxr-xr-x 1 root root    21035 30. Nov 06:04  bcm2837-rpi-3-b.dtb
-rwxr-xr-x 1 root root    21499 30. Nov 06:04  bcm2837-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root    20174 30. Nov 06:04  bcm2837-rpi-cm3-io3.dtb
-rwxr-xr-x 1 root root    20600 30. Nov 06:04  bcm2837-rpi-zero-2-w.dtb
-rwxr-xr-x 1 root root    52476 30. Nov 06:04  bootcode.bin
-rwxr-xr-x 1 root root      180 24. Sep 21:25  cmdline.txt
-rwxr-xr-x 1 root root     1239 24. Sep 21:25  config.txt
-rwxr-xr-x 1 root root     3204 30. Nov 06:04  fixup4cd.dat
-rwxr-xr-x 1 root root     5434 30. Nov 06:04  fixup4.dat
-rwxr-xr-x 1 root root     8423 30. Nov 06:04  fixup4db.dat
-rwxr-xr-x 1 root root     8425 30. Nov 06:04  fixup4x.dat
-rwxr-xr-x 1 root root     3204 30. Nov 06:04  fixup_cd.dat
-rwxr-xr-x 1 root root     7303 30. Nov 06:04  fixup.dat
-rwxr-xr-x 1 root root    10268 30. Nov 06:04  fixup_db.dat
-rwxr-xr-x 1 root root    10268 30. Nov 06:04  fixup_x.dat
-rwxr-xr-x 1 root root 12724469 30. Nov 06:04  initrd.img
drwxr-xr-x 2 root root    29696 30. Nov 06:04  overlays
-rwxr-xr-x 1 root root   808892 30. Nov 06:04  start4cd.elf
-rwxr-xr-x 1 root root  3753480 30. Nov 06:04  start4db.elf
-rwxr-xr-x 1 root root  2256224 30. Nov 06:04  start4.elf
-rwxr-xr-x 1 root root  3004040 30. Nov 06:04  start4x.elf
-rwxr-xr-x 1 root root   808892 30. Nov 06:04  start_cd.elf
-rwxr-xr-x 1 root root  4825352 30. Nov 06:04  start_db.elf
-rwxr-xr-x 1 root root  2980544 30. Nov 06:04  start.elf
-rwxr-xr-x 1 root root  3727656 30. Nov 06:04  start_x.elf
drwxr-xr-x 2 root root      512 30. Okt 10:53 'System Volume Information'
-rwxr-xr-x 1 root root 25534976 30. Nov 06:04  vmlinuz
pi@ncpi4:~$
```

fhamborg, 2024-12-05:
cool thanks a lot! after some diving into this, i found that for some reason the wrong kernel is configured for my system to update. 

```
cat /etc/apt/sources.list
deb http://deb.debian.org/debian bookworm main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm main contrib non-free non-free-firmware

deb http://deb.debian.org/debian bookworm-updates main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm-updates main contrib non-free non-free-firmware

deb http://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware

deb http://security.debian.org/ bookworm-security main contrib non-free non-free-firmware
#deb-src http://security.debian.org/ bookworm-security main contrib non-free non-free-firmware
```

```
cat /etc/apt/sources.list.d/*.list
deb [signed-by=/usr/share/keyrings/armbian.gpg] http://apt.armbian.com bookworm main bookworm-utils bookworm-desktop
deb [arch=arm64 signed-by=/etc/apt/keyrings/docker.asc] https://download.docker.com/linux/debian bookworm stable
deb https://packages.sury.org/php/ bookworm main
```

due to this, the debian bookworm kerneles are being pulled (see first output above) as well as the (correct) armbian ones (second above). i made some changes to prioritize armbian kernels higher than debian kernels and now it seems to work. this is what i added:

```
cat /etc/apt/preferences.d/armbian
#Explanation: Uninstall or do not install chromium, thunderbird, firefox Ubuntu-originated
#Explanation: package versions other than those in the Armbian distro
#Package: chromium thunderbird firefox
#Pin: release o=armbian
#Pin-Priority: 990
#
#Package: chromium thunderbird firefox
#Pin: release o=Ubuntu
#Pin-Priority: 50
#

Package: *
Pin: origin apt.armbian.com
Pin-Priority: 1001
```

and 

```
cat /etc/apt/preferences.d/backports
Package: *
Pin: release n=bookworm-backports
Pin-Priority: 100
```

in case you have time, would be interesting to see if you also have any of these two files set or if not im really wondering why your system pulls the correct one (from both sources) whereas mine doesnt.

geoW-n, 2024-12-05:
ncpi4 is my test instance, fresh install with latest ncp 155.3 (armbian based)

```
pi@ncpi4:~$ cat /etc/apt/preferences.d/armbian
#Explanation: Uninstall or do not install chromium, thunderbird, firefox Ubuntu-originated
#Explanation: package versions other than those in the Armbian distro
#Package: chromium thunderbird firefox
#Pin: release o=armbian
#Pin-Priority: 990
#
#Package: chromium thunderbird firefox
#Pin: release o=Ubuntu
#Pin-Priority: 50
```
```
pi@ncpi4:~$ cat /etc/apt/preferences.d/backports
cat: /etc/apt/preferences.d/backports: Datei oder Verzeichnis nicht gefunden
pi@ncpi4:~$
```

nextcloudpi is my production instance, updatet to ncp 155.3 (stays debian based)
```
pi@nextcloudpi:~ $ cat /etc/apt/preferences.d/armbian
cat: /etc/apt/preferences.d/armbian: Datei oder Verzeichnis nicht gefunden
pi@nextcloudpi:~ $
```
```
pi@nextcloudpi:~ $ cat /etc/apt/preferences.d/backports
cat: /etc/apt/preferences.d/backports: Datei oder Verzeichnis nicht gefunden
pi@nextcloudpi:~ $
```

fhamborg, 2024-12-05:
Ah weird, and what's the content when you use the two cat commands (the first two, with sources)?

geoW-n, 2024-12-05:
ncpi4
```
pi@ncpi4:~$ cat /etc/apt/sources.list
deb http://deb.debian.org/debian bookworm main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm main contrib non-free non-free-firmware

deb http://deb.debian.org/debian bookworm-updates main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm-updates main contrib non-free non-free-firmware

deb http://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware

deb http://security.debian.org/ bookworm-security main contrib non-free non-free-firmware
#deb-src http://security.debian.org/ bookworm-security main contrib non-free non-free-firmware
pi@ncpi4:~$
```
```
pi@ncpi4:~$ cat /etc/apt/sources.list.d/*.list
deb [signed-by=/usr/share/keyrings/armbian.gpg] https://github.armbian.com/configng stable main
deb [signed-by=/usr/share/keyrings/armbian.gpg] http://apt.armbian.com bookworm main bookworm-utils bookworm-desktop
deb https://packages.sury.org/php/ bookworm main
pi@ncpi4:~$
```
nextcloudpi
```
pi@nextcloudpi:~ $ cat /etc/apt/sources.list
deb http://deb.debian.org/debian bookworm main contrib non-free
deb http://security.debian.org/debian-security bookworm-security main contrib non-free
deb http://deb.debian.org/debian bookworm-updates main contrib non-free
# Uncomment deb-src lines below then 'apt-get update' to enable 'apt-get source'
#deb-src http://deb.debian.org/debian bookworm main contrib non-free
#deb-src http://security.debian.org/debian-security bookworm-security main contrib non-free
#deb-src http://deb.debian.org/debian bookworm-updates main contrib non-free
pi@nextcloudpi:~ $
```
```
pi@nextcloudpi:~ $ cat /etc/apt/sources.list.d/*.list
deb https://packages.sury.org/php/ bookworm main
deb http://archive.raspberrypi.org/debian/ bullseye main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ bullseye main
pi@nextcloudpi:~ $

```

fhamborg, 2024-12-05:
Okay that's interesting, you also got the Debian bookworm source (from which the wrong kernel I always ended up having after performing an update or having an auto update).

fhamborg, 2024-12-05:
Anyway, thanks for your help :)

geoW-n, 2024-12-05:
If I read your first post right, you use a pi5?
I just started a ncpi5 test instance ncp1.55.3, here the output of the cat commands.

ncpi5
```
pi@ncpi5:~$ cat /etc/apt/preferences.d/armbian
#Explanation: Uninstall or do not install chromium, thunderbird, firefox Ubuntu-originated
#Explanation: package versions other than those in the Armbian distro
#Package: chromium thunderbird firefox
#Pin: release o=armbian
#Pin-Priority: 990
#
#Package: chromium thunderbird firefox
#Pin: release o=Ubuntu
#Pin-Priority: 50
```
```
pi@ncpi5:~$ cat /etc/apt/preferences.d/backports
cat: /etc/apt/preferences.d/backports: Datei oder Verzeichnis nicht gefunden
pi@ncpi5:~$
```
```
pi@ncpi5:~$ cat /etc/apt/sources.list
deb http://deb.debian.org/debian bookworm main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm main contrib non-free non-free-firmware

deb http://deb.debian.org/debian bookworm-updates main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm-updates main contrib non-free non-free-firmware

deb http://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware

deb http://security.debian.org/ bookworm-security main contrib non-free non-free-firmware
#deb-src http://security.debian.org/ bookworm-security main contrib non-free non-free-firmware
pi@ncpi5:~$
```
```
pi@ncpi5:~$ cat /etc/apt/sources.list.d/*.list
deb [signed-by=/usr/share/keyrings/armbian.gpg] https://github.armbian.com/configng stable main
deb [signed-by=/usr/share/keyrings/armbian.gpg] http://apt.armbian.com bookworm main bookworm-utils bookworm-desktop
deb https://packages.sury.org/php/ bookworm main
pi@ncpi5:~$
```

fhamborg, 2024-12-06:
yes, im using a pi 5. your sources look pretty identical to mine i think. so the reason seems to be on my system that for some reason the wrong kernel was pulled.