nextcloud/nextcloudpi #2019: Installing Nextcloud 30.0.3 failde

Schniefel, 2024-12-06:
I can't install nextcloud 30.0.3. No matter if I update to version '0' or '30.0.3' I get the following console output:

Download Nextcloud 30.0.3...
Error downloading
Clean up...
Nextcloud update failed (or was installed already).
Done. Press any key...

geoW-n, 2024-12-07:
> I can't install nextcloud 30.0.3

You are howling the wrong moon, it is an upstream issue.
Take a look:
[https://help.nextcloud.com/t/28-0-13-and-30-0-3-withdrawn/211672](https://help.nextcloud.com/t/28-0-13-and-30-0-3-withdrawn/211672)

Schniefel, 2024-12-07:
thank you, good to know

Sennat0r, 2025-01-15:
It works now, I've successfully updated from NC v29.0.10 to v30.0.4 on my NCP v1.55.3 2 weeks ago.

closed 2025-02-04 by joshtrichards