nextcloud/nextcloudpi #2034: Option to disable HTTP2 when behind external proxy

MB-Finski, 2025-02-03:
**Description**  

An option to disable HTTP2 (h2,h2c) would be desirable. Currently the apache2 conf files are overwritten so it's not possible to disable http2 without disabling some of the nextcloudpi scripts. 

HTTP2 causes issues when nextcloudpi is behind a NGINX reverse proxy because nextcloudpi will return "Upgrade: h2,h2c" header for every request. This will cause mobile Safari clients to fail to connect to Nextcloudpi altogether. On the other hand, if the upgrade header is removed, notify_push websocket connection will no longer work since in this instance the "Upgrade: websocket" header will be removed.

The easiest solution would be to just disable HTTP2 altogether since NGINX does not support it in any case.

RapidRotator, 2025-03-03:
In the NextcloudPi panel, there is an option called nc-httpsonly that appears to allow this, but it doesn't seem to have any effect. Did I misunderstand the option's function or is this now a bug?

RapidRotator, 2025-03-03:
Rebooting my RaspberryPI with nextcloudpi installed has removed the upgrade header. @MB-Finski can you confirm?

RapidRotator, 2025-03-03:
I second this.

einnicernutzer, 2025-03-19:
i have my nextcloud instance connected to the cloudflare zero tier proxy and i cant upload / download anything which is kind of a bummer, bc i really would like to get away from the whole google stuff. But i also dont want to expose my home network to the internet.