nextcloud/nextcloudpi #2039: Protect better agains accidential nc-init

janvlug, 2025-03-12:
**Description**  
Let the user extra confirm the nc-init operation.

**Example**  
Maybe by asking to type something that is unique (i.e. is not saved in the browser auto fill in). Or maybe make the Apply button the color red. But basically, with auto filling on in the browser you are a few clicks away from re-initializing your nextcloudpi (and thus potentially losing data).

**Background of this request**
I accidentally did an nc-init on my nextcloudpi. This happened because there was some difficulty in updating the letsencrypt ssl certificate, because port 80 was not open.
I wanted to disable https-only. the nc-https-only is just above nc-init:

[image]

Accidentally, instead of on nc-https-only, I clicked on nc-init.
I have automatic filling in of the password enabled for many sites, so username and password were pre-filled in:

[image]

Because I had been fiddling already a while with settings to try to get the certificate renewed, I did not read the message, but just hit Apply. I agree that there is a clear warning in capitals, but because I went so many times through the screens and clicking Apply previously, it was apparently not enough to grap my attention, and I still accidentally clicked on Apply. Only when I saw the output messages appearing I realized my mistake, that I could not stop or undo any more.

janvlug, 2025-03-12:
Another option might be to make automatically a backup of the `nextcloud` database before dropping it.

C-Fu, 2025-03-13:
Agreed! For something that can be a huge disaster it's too inconspicuous to be hidden in the middle like any other option.
just <color=red> tag at the left menu name is a huge difference

FadeFx, 2025-03-23:
in case someone is doing something about this, please consider the same for nc-limits, which iirc will also reset Nextcloud.