nextcloud/nextcloudpi #2040: Internal Server Error, cannot access NCP dashboard/files

bluejay76, 2025-03-14:
I've run the steps in #1913 and still cannot use my NCP (since last 6 months or so; usage broke without my intervention, with some background update). I get an Internal Server Error and unable to load the dashboard. I can access Panel, which shows all services/checks ok/up. PHP I believe used to be down, but at least that appears fixed now. (I downgraded to version 8.1). I'm on v1.55.3, image NextCloudPi_10-08-21. Have even been using AI to diagnose and help with fixing to no avail. Can anyone help?

I get this in the Panel log after reboot, unsure if it's important: read error on connection to /var/run/redis/redis.sock:0.
Command: `systemctl status redis.service` check shows the service as active.

NCP report attached

[NCP report.txt](https://github.com/user-attachments/files/19254008/NCP.report.txt)

bluejay76, 2025-03-15:
I also keep finding that NC is upgrading to a PHP version unsupported by NC: pi@nextcloudpi:~ $ sudo ncc maintenance:mode --off
This version of Nextcloud is not compatible with PHP>=8.4.<br/>You are currently running 8.4.5.

I had just yesterday manually set PHP to version 8.1 to try to circumvent any compatibility issues, but today it's an even later version than prior to that manual change (must have auto-updated in the past 24 hours).

sebeweiss, 2025-03-15:
Do you use `'host' => '/var/run/redis/redis.sock'` in your `config.php`?

This is what helped me with a similar, but probably not the same problem:
```
sudo -u www-data php /var/www/nextcloud/occ notify_push:setup
```

bluejay76, 2025-03-16:
Thanks for the tip. It gave the message below. I'm not an expert, so I'm not sure if my setup qualifies as a single server instance, running on a pi with USB-attached hard drive. Is your system similar (so that I at least know it didn't brick anything for you)?

This setup wizard is intended for use on single server instances
where the nextcloud server, web server/reverse proxy and push daemon all run on the same machine.
If your setup is more complex or involves any kind of load balancing
you should follow the manual setup instruction on the README instead

> Do you use `'host' => '/var/run/redis/redis.sock'` in your `config.php`?
> 
> This is what helped me with a similar, but probably not the same problem:
> 
> ```
> sudo -u www-data php /var/www/nextcloud/occ notify_push:setup
> ```

sebeweiss, 2025-03-17:
Yes, it is similar. But it's never a bad idea to make a backup.
By the way, my PHP version is 8.3.

FadeFx, 2025-03-23:
afaik PHP 8.1 is no more supported from Nextcloud 29 on. you most liekly need to install 8.3 and dont forget to install FPM and all other dependencies for Nextcloud.

probably this walkthough (especially step 2) may be of help here:
https://ardadanaci.com/en/blog/nextcloud