nextcloud/nextcloudpi #2046: Not possible to update to 1.55.4

DeineBla, 2025-04-04:
Cannot update from 1.55.3 to 1.55.4. 

Strange is, that nextcloud was automatically updated to version 31.

Also the 1.55.4 version cannot be downloaded: https://github.com/nextcloud/nextcloudpi/releases

**System information**

NextcloudPi version  v1.55.3
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.0.img
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda sdb sdc 
datadir              /media/SSD/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      663G/1.9T
rootfs usage         15G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    31.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               22:21

```
</details>

TomF79, 2025-04-11:
@theCalcaholic 

Is it possible that anything went wrong with the Release from 1.55.4 three weeks ago?

BR Tom

theCalcaholic, 2025-04-11:
There is no version 1.55.4 yet, due to an issue with the build pipeline.
Updates to Nextcloud 31 are still supported already.

closed 2025-04-11 by theCalcaholic

TomF79, 2025-04-11:
> There is no version 1.55.4 yet, due to an issue with the build pipeline. Updates to Nextcloud 31 are still supported already.

@theCalcaholic 

In the Panel is 30.0.9 the latest. Is this correct?

[image]

Adnapeht, 2025-04-11:
> There is no version 1.55.4 yet, due to an issue with the build pipeline. Updates to Nextcloud 31 are still supported already.

so does this mean that version 1.55.4 will be rolled out sometime soon? I'm setting up my nextcloud right now, but the temp folder bug from 1.55.3 is halting my progress :/

theCalcaholic, 2025-04-11:
> > There is no version 1.55.4 yet, due to an issue with the build pipeline. Updates to Nextcloud 31 are still supported already.
> 
> @theCalcaholic so does this mean that version 1.55.4 will be rolled out sometime soon? I'm setting up my nextcloudpi right now, but the temp folder bug from 1.55.3 is halting my progress :/
> 
> edit: if 1.55.4 is not yet available, how would i go about regression to a previous version to circumnavigate the bug? I'm not super experienced with linux, but i know how to access the ncp dashbord and SSH, so a simple solution (even if that's just to wait for the new version) is preferred :)

Yes, you can manually run `ncp-update` today to update to the latest (future 1.55.4) version.

Adnapeht, 2025-04-11:
SSH'd in and ran ncp-update, but it looks like it didn't update, staying on v1.55.3. From what you said it sounded like 1.55.4 was the current most recent version, but did I misinterpret your response? Like I said, I don't mind waiting until the version is released, but I also don't want to wait if this is something I can fix now.

TomF79, 2025-04-12:
> Yes, you can manually run `ncp-update` today to update to the latest (future 1.55.4) version.

Now I understand.

I have updated ncp again to 1.55.3 and then nextcloud update to 31.0.3 appears to install.

Adnapeht, 2025-04-14:
@theCalcaholic Ohhhh, that makes sense now why the bug went away seemingly inexplicably. Thank you so much for your help and your replies, they're much appreciated :)