nextcloud/nextcloudpi #2047: No updates and installations possible

blueskyfeeling, 2025-04-12:
Hello, 
I have installed version NextcloudPi_RaspberryPi4_v1.55.3 on my Raspberry 4B. 

Unfortunately, an update or a new installation of another program no longer works. I get the following error message. 

[image]

apt clean, apt autoremove and also "apt --fix-broken install" didn't help.

I ask for your support.

geoW-n, 2025-04-12:
You may try to delete 
`libraspberrypi0`
`libraspberrypi0-bin`

Be careful to install other software, keep in mind ncp now is armbian based, if not installed install 
`armbian-config`
to upgrade os use
`armbian-upgrade`

blueskyfeeling, 2025-04-12:
Attached are my search results. Which files should I delete specifically?

[image]

geoW-n, 2025-04-12:
I can't say what files you should delete.
But I tell you how I managed the error message on question on my rpi4 test server.
`apt remove libraspberrypi0 libraspberrypi-bin`
For me it allows to finish an armbian-upgrade command without errors but with warnings.

blueskyfeeling, 2025-04-13:
I can currently only access this Raspberry Pi remotely. If I follow your suggestion and the device stops working, that wouldn't be a good idea. Or is it safe?

geoW-n, 2025-04-13:
Come on, no one will give you a warranty for a server you set up.
I told you my experience with this armbian raspi firmware issue, which quirks the packet management. you may read about more on armbian forum.
Your server is up and running, no need to change it, Stop auto updating and set up a sibling server you may check the commands you will send to your working server.

rummu666, 2025-04-13:
removed both packages and all good again.
`apt remove libraspberrypi0 libraspberrypi-bin`

blueskyfeeling, 2025-04-13:
Well, I've done it now. And everything works. Thanks everyone!

heinovski, 2025-04-14:
This issue seems to be related to / cause by https://github.com/armbian/build/issues/8069.
A corresponding PR has already been created https://github.com/armbian/build/pull/8087.