nextcloud/nextcloudpi #2056: Installation fails with ERROR 1146 (42S02) at line 1: Table 'nextcloud.appconfig' doesn't exist

noxamox23, 2025-05-18:
Hi,
just tried to install ncp on my raspi using 
curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh|sudo bash

The script seems to have run through most of it but ended with an error message, I copy here the end of the output:

[...]
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory set to string /var/www/nextcloud/data/tmp
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value trusted_domains => 8 set to string nextcloudpi.lan
System config value trusted_domains => 3 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string noreply
System config value mail_domain set to string nextcloudpi.com
ERROR 1146 (42S02) at line 1: Table 'nextcloud.appconfig' doesn't exist

I tried to trace the location where this occurred - it seems to be somewhere in nc-init.sh.

I also tried to find out how to fix this. I manually ran ncp-config and had run nc-init which went through without an error.
So in addition to reporting this issue it would be nice to know whether I can use the installation and how to start it up safely - if this question should be asked in a forum instead please kindly point me to the right location...

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version                                                                           v1.55.4
NextcloudPi image                                                                             NextcloudPi_05-18-25
OS                                                                                            Raspbian GNU/Linux 12
NCP is not activated yet. Please enter https://nextcloudpi.local or this instance's local IP  ***REMOVED SENSITIVE VALUE***
automount                                                                                     no
USB devices                                                                                   sda 
datadir                                                                                       /var/www/nextcloud/data
data in SD                                                                                    yes
data filesystem                                                                               ext2/ext3
data disk usage                                                                               8.7G/29G
rootfs usage                                                                                  8.7G/29G
swapfile                                                                                      /var/swap
dbdir                                                                                         /var/lib/mysql
Nextcloud check                                                                               ok
Nextcloud version                                                                             31.0.2.1
HTTPD service                                                                                 up
PHP service                                                                                   up
MariaDB service                                                                               up
Redis service                                                                                 up
HPB service                                                                                   down
Postfix service                                                                               up
Internet check                                                                                ok
Public IPv4                                                                                   ***REMOVED SENSITIVE VALUE***
Public IPv6                                                                                   ***REMOVED SENSITIVE VALUE***
Port 80                                                                                       closed
Port 443                                                                                      closed
IP                                                                                            ***REMOVED SENSITIVE VALUE***
Gateway                                                                                       ***REMOVED SENSITIVE VALUE***
Interface                                                                                     wlan0
Certificates                                                                                  ***REMOVED SENSITIVE VALUE***
NAT loopback                                                                                  no
Uptime                                                                                        14days

```
</details>

<details>
[... 77 lines cut ...]

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"2HK17PnswK6h9kQiM0BZ","level":3,"time":"2025-05-18T09:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":691,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1156,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":24,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":228,"message":"Not installed","exception":{},"CustomMessage":"Not installed"}}
{"reqId":"rjFWZgU7AE886hFR91My","level":3,"time":"2025-05-18T09:05:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":691,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1156,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":24,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":228,"message":"Not installed","exception":{},"CustomMessage":"Not installed"}}
{"reqId":"krklL02zkYiIOyRcwuP8","level":2,"time":"2025-05-18T10:05:02+00:00","remoteAddr":"","user":false,"app":"cron","method":"","url":"--","message":"failed to create instance of background job: OCA\\UpdateNotification\\BackgroundJob\\UpdateAvailableNotifications","userAgent":"--","version":"31.0.2.1","exception":{"Exception":"OC\\AppFramework\\Utility\\QueryNotFoundException","Message":"Could not resolve OCA\\UpdateNotification\\BackgroundJob\\UpdateAvailableNotifications! Class \"OCA\\UpdateNotification\\BackgroundJob\\UpdateAvailableNotifications\" does not exist","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":124,"function":"resolve","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/ServerContainer.php","line":155,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":42,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/var/www/nextcloud/lib/public/Server.php","line":39,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/JobList.php","line":288,"function":"get","class":"OCP\\Server","type":"::"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/JobList.php","line":197,"function":"buildJob","class":"OC\\BackgroundJob\\JobList","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":152,"function":"getNext","class":"OC\\BackgroundJob\\JobList","type":"->"}],"File":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","Line":113,"message":"failed to create instance of background job: OCA\\UpdateNotification\\BackgroundJob\\UpdateAvailableNotifications","exception":{},"CustomMessage":"failed to create instance of background job: OCA\\UpdateNotification\\BackgroundJob\\UpdateAvailableNotifications"}}

```
</details>

geoW-n, 2025-05-21:
Yes, I think you are better off on help forum with ncp mark.
I never used the curl script.
What rpi are you running?
You are running Raspian Os, updated?
It seems you have to activate your instance, https://nextcloudpi[.lan | .local] or https://\<ip-number\> .

MahrWe, 2025-06-19:
I encountered the same issue on Armbian Bookworm.