nextcloud/nextcloudpi #2066: Fugitive "System info" and blank page with message "Something went wrong. Try refreshing the page" since V1.55.4

dwprod38, 2025-09-09:
After Wifi drivers missing on Raspberry Pi 5 with Armbian version 25.2.3 solved: I decide to publish because no more news from developers about issue discovered since 1.55.4 update... [Link](https://github.com/nextcloud/nextcloudpi/issues/2055)

NextcloudPi panel V 1.55.4: after reloading page or clicking different functionality from left side menu, few seconds the panel display a blank page with message "Something went wrong. Try refreshing the page". Despite refreshing same page, the blank page still back after few seconds with same message... This issue appeared after updating from 1.54.0 to 1.55.4. --> I've tested same version on RPI4 Debian OS and v 1.55.4: works fine and stable.

See video example: https://youtu.be/dDv5FH03gjA

I think that bug is linked with "Nextcloudpi 1.55.4" running on RPI5 despite after armbian update to version 25.8.1, nothing changed but more reliable and faster than previous version 25.2.3

**System information**

NextcloudPi version v1.55.4
NextcloudPi image
OS Armbian 25.8.1 bookworm \l . 6.6.63-current-bcm2712 (aarch64)
automount yes
USB devices none
datadir REMOVED SENSITIVE VALUE
data in SD no
data filesystem btrfs
data disk usage 2.1T/3.7T
rootfs usage 7.5G/58G
swapfile /var/swap
dbdir REMOVED SENSITIVE VALUE
Nextcloud check ok
Nextcloud version 31.0.8.1
HTTPD service up
PHP service up
MariaDB service up
Redis service up
HPB service up
Postfix service up
Internet check ok
Public IPv4 REMOVED SENSITIVE VALUE
Public IPv6 REMOVED SENSITIVE VALUE
Port 80 open
Port 443 open
IP REMOVED SENSITIVE VALUE
Gateway REMOVED SENSITIVE VALUE
Interface end0
Certificates REMOVED SENSITIVE VALUE
NAT loopback no
Uptime 12:17

Nextcloud configuration
{
"system": {
"passwordsalt": "REMOVED SENSITIVE VALUE",
"secret": "REMOVED SENSITIVE VALUE",
"trusted_domains": {
"0": "localhost",
"3": "nextcloudpiclone",
"7": "nextcloudpiclone",
"5": "nextcloudpiclone.local",
"8": "nextcloudpiclone.lan",
"1": "192.168.2.247",
"14": "nextcloudpiclone"
},
"datadirectory": "REMOVED SENSITIVE VALUE",
"dbtype": "mysql",
"version": "31.0.8.1",
"overwrite.cli.url": "REMOVED SENSITIVE VALUE",
"dbname": "REMOVED SENSITIVE VALUE",
"dbhost": "REMOVED SENSITIVE VALUE",
"dbport": "",
"dbtableprefix": "oc_",
"mysql.utf8mb4": true,
"dbuser": "REMOVED SENSITIVE VALUE",
"dbpassword": "REMOVED SENSITIVE VALUE",
"installed": true,
"instanceid": "REMOVED SENSITIVE VALUE",
"memcache.local": "\OC\Memcache\Redis",
"memcache.locking": "\OC\Memcache\Redis",
"redis": {
"host": "REMOVED SENSITIVE VALUE",
"port": 0,
"timeout": 0,
"password": "REMOVED SENSITIVE VALUE"
},
"tempdirectory": "REMOVED SENSITIVE VALUE",
"mail_smtpmode": "sendmail",
"mail_smtpauthtype": "LOGIN",
"mail_from_address": "REMOVED SENSITIVE VALUE",
"mail_domain": "REMOVED SENSITIVE VALUE",
"preview_max_x": "2048",
"preview_max_y": "2048",
"jpeg_quality": "60",
"overwriteprotocol": "https",
"default_phone_region": "FR",
"maintenance_window_start": 1,
"maintenance": false,
"trusted_proxies": "REMOVED SENSITIVE VALUE",
"logfile": "REMOVED SENSITIVE VALUE",
"data-fingerprint": "REMOVED SENSITIVE VALUE",
"theme": "",
"loglevel": 2
}
}

dwprod38, 2025-09-10:
hello @theCalcaholic, 

Please have look about this specific issue with RPI5+armbian+1.55.4 before publishing next officlal nextcloudpi version in GitHub ;)

markusonwheels83, 2026-04-14:
same on 1.57

NeoCoderMatrix86, 2026-07-14:
If it helps, I can also help to investigate this issue. On my system the system-info also stays empty, while the output of the bash works.

dwprod38, 2026-08-25:
Dear users, the bug is identified :

After checking each "ncp-launcher.php" file in Chrome inspector/Network/Fetch-XHR, I found the content inside one of them:
"error opening /var/www/ncp-web/backup-info-cache.cfg"

I've checked the files suspicious permissions with this command:
_sudo ls -la /var/www/ncp-web/backup-info-cache.cfg_

Everyting is ok with permissions
_-rw-r--r-- 1 www-data www-data 0 Apr 20  2025 /var/www/ncp-web/backup-info-cache.cfg_

But this file is 0 bytes (empty). In PHP, file_get_contents() on an empty file returns an empty string "" — which PHP treats as "falsy". In the lines 54 & 55 of "backups.php" script (located in "/var/www/ncp-web") :

_$cache_str = file_get_contents($cache_file)
  or exit("error opening ${cache_file}");_

Because of how PHP's or operator works (very low precedence), this behaves like:

_($cache_str = file_get_contents($cache_file)) or exit(...)_

So even though the file read technically succeeds, since the result is an empty string (""), PHP treats it as a failure and still triggers exit(). This is a logic bug in the "backups.php" script: it should have checked file_get_contents($cache_file) !== false instead of relying on the "truthiness" of the file's content. Something have slight difference PHP between Armbian/RPI5 package and Debian/RPI4 package.

So I invite you to try the fix on RPI5 users to put valid, non-empty JSON content in this file — for example an empty JSON object {}:

_echo '{}' | sudo tee /var/www/ncp-web/backup-info-cache.cfg
sudo chown www-data:www-data /var/www/ncp-web/backup-info-cache.cfg
sudo chmod 664 /var/www/ncp-web/backup-info-cache.cfg_

Then reload "System info" in the browser — it should work like as me.

Regards

markusonwheels83, 2026-09-22:
sadly doesn't help on my debian13 x86_64 vm on proxmox. Bug still exist's.