nextcloud/nextcloudpi #2067: upgrade nc from 28.0.14 to 29.0.16.1 // ERROR 1146 (42S02) 'nextcloud.oc_jobs' doesn't exist

lfriedri, 2025-09-14:
Starting from a working ncp system with

ncp version 1.55.4
nc version 28.0.14

I used the ncp web frontend to run `nc-update-nextcloud`.
Now, the nc-server can not be reached from clients.

Running `sudo ncp-update-nc 0 `gives
ERROR 1146 (42S02) at line 1: Table 'nextcloud.oc_jobs' doesn't exist

Running `sudo -u www-data php occ maintenance:repair`
does not resolve the issue but reports errors about missing tables.

[ncp-report.txt](https://github.com/user-attachments/files/22317929/ncp-report.txt)

lfriedri, 2025-09-14:
might be related to #1818

theCalcaholic, 2025-09-30:
Hi, thanks for reporting. Can you provide the output of:

```bash
sudo mysql -D nextcloud <<<'SHOW TABLES;'
```

theCalcaholic, 2025-10-02:
Alright, sorry I couldn't help earlier

closed 2025-10-02 by theCalcaholic