nextcloud/nextcloudpi #2072: HSTS is not configurable in the GUI

hubertmeier, 2025-09-27:
i built a nextcloupi and use only a portforwading 443 -> 8445 in the router, i do not use letsencrypt or so. The domain

xxxxx.myfritz.net

is added to the trusted-domains. Its not really clear, if there must be an entry like xxxxx.myfritz.net:8445 oder without the port. Further the hostname nc5 is added too ...

Obviously is HSTS activated, the clinet 3.17. does not accept the certificate, the browsers work without problems. There is absolutely no hint in the GUI that HSTS exists and the status, activated or not or the possibility to change that.

So i can use only Client 3.16. or earlier to access the nextcloud. The same problem appeared with an older nextcloudpi which worked for years after the clinet update to 3.17.

I think, many many nextcloud instances run like this.

hubertmeier, 2025-10-07:
Port 8445 is the extern forwarded port 443 from the Nextcloudpi

Xxxxx.myfritz.net:8445 -> 192.x.x.x:443

For an older Nextcloudpi I use extern 8444

hubertmeier, 2025-10-08:
Hi again, disabling HSTS by commenting out the header in the file /etc/apache2/sites-available/001-nextcloud.conf is obviously not enough to get the desktop client 3.17. to work. The header is really not sent now, but the client does not accept a self signed certificate.

The changes made in client 3.17. are not really clear, they stated that HSTS is the problem (its not a bug, its a security feature), but with deactivated HSTS-Header the problem remains. The acces per browser works (unsecure, with accepting the self signed certificate). So we need a nextcloupi-configuration which works with self-signed certificates and forwarded ports. Is it possible that the portforwarding (intern 443, extern 8445) causes these problems?

tschenkelz, 2025-12-25:
Hello.

I use nextcloud AIO and I dont see a reasonable way to deal with this issue.
I cannot connect to my nextcloud with my nextcloud desktop app anymore.

I use my own CA at my nextcloud. Worked before, now I cant get it back to work.